jsk-ros-pkg/coral_usb_ros #1: Cannot build a workspace on README.md

sktometometo, 2019-09-15:
I had tried to build the package following the instruction in README.md, but "catkin build" command had failed because "catkin_pkg" was not found during "catkin_tools_prebuild" process.

```shell
sktometometo@Sherman ~/ros/ws_tpu_development/src [gitbranch:] [workspace:] ->
 06:28 午後 日  9月 15$ catkin clean

[clean] Warning: This will completely remove the following directories. (Use `--yes` to skip this check)
[clean] Log Space:     /home/sktometometo/ros/ws_tpu_development/logs
[clean] Build Space:   /home/sktometometo/ros/ws_tpu_development/build
[clean] Devel Space:   /home/sktometometo/ros/ws_tpu_development/devel

[clean] Are you sure you want to completely remove the directories listed above? [yN]: y
[clean] Removing develspace: /home/sktometometo/ros/ws_tpu_development/devel
[clean] Removing buildspace: /home/sktometometo/ros/ws_tpu_development/build
[clean] Removing log space: /home/sktometometo/ros/ws_tpu_development/logs
sktometometo@Sherman ~/ros/ws_tpu_development/src [gitbranch:] [workspace:] ->
 06:28 午後 日  9月 15$ cd ..
sktometometo@Sherman ~/ros/ws_tpu_development [gitbranch:] [workspace:] ->
 06:28 午後 日  9月 15$ ls
src
sktometometo@Sherman ~/ros/ws_tpu_development [gitbranch:] [workspace:] ->
 06:28 午後 日  9月 15$ ls -a
.  ..  .catkin_tools  src
sktometometo@Sherman ~/ros/ws_tpu_development [gitbranch:] [workspace:] ->
 06:28 午後 日  9月 15$ rm -rf .catkin_tools/
sktometometo@Sherman ~/ros/ws_tpu_development [gitbranch:] [workspace:] ->
 06:28 午後 日  9月 15$ catkin init
Initializing catkin workspace in `/home/sktometometo/ros/ws_tpu_development`.
------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/sktometometo/ros/ws_tpu_development
------------------------------------------------------------------------------
Build Space:       [missing] /home/sktometometo/ros/ws_tpu_development/build
Devel Space:       [missing] /home/sktometometo/ros/ws_tpu_development/devel
Install Space:      [unused] /home/sktometometo/ros/ws_tpu_development/install
Log Space:         [missing] /home/sktometometo/ros/ws_tpu_development/logs
Source Space:       [exists] /home/sktometometo/ros/ws_tpu_development/src
DESTDIR:            [unused] None
------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
------------------------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------------------------
sktometometo@Sherman ~/ros/ws_tpu_development [gitbranch:] [workspace:] ->
 06:28 午後 日  9月 15$ catkin config -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.5m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.5m.so
---------------------------------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/sktometometo/ros/ws_tpu_development
---------------------------------------------------------------------------------------------------------
Build Space:       [missing] /home/sktometometo/ros/ws_tpu_development/build
Devel Space:       [missing] /home/sktometometo/ros/ws_tpu_development/devel
Install Space:      [unused] /home/sktometometo/ros/ws_tpu_development/install
Log Space:         [missing] /home/sktometometo/ros/ws_tpu_development/logs
Source Space:       [exists] /home/sktometometo/ros/ws_tpu_development/src
DESTDIR:            [unused] None
---------------------------------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
[... 220 lines cut ...]
  ros-kinetic-rqt-py-console ros-kinetic-rqt-reconfigure ros-kinetic-rqt-robot-dashboard
  ros-kinetic-rqt-robot-monitor ros-kinetic-rqt-robot-plugins ros-kinetic-rqt-robot-steering
  ros-kinetic-rqt-runtime-monitor ros-kinetic-rqt-rviz ros-kinetic-rqt-service-caller
  ros-kinetic-rqt-shell ros-kinetic-rqt-srv ros-kinetic-rqt-tf-tree ros-kinetic-rqt-top
  ros-kinetic-rqt-topic ros-kinetic-rqt-web ros-kinetic-rviz ros-kinetic-rviz-plugin-tutorials
  ros-kinetic-rviz-python-tutorial ros-kinetic-self-test ros-kinetic-sensor-msgs ros-kinetic-shape-msgs
  ros-kinetic-simulators ros-kinetic-smach-msgs ros-kinetic-smach-ros ros-kinetic-sound-play
  ros-kinetic-stage ros-kinetic-stage-ros ros-kinetic-std-msgs ros-kinetic-std-srvs
  ros-kinetic-stereo-image-proc ros-kinetic-stereo-msgs ros-kinetic-tf ros-kinetic-tf-conversions
  ros-kinetic-tf2 ros-kinetic-tf2-eigen ros-kinetic-tf2-geometry-msgs ros-kinetic-tf2-kdl
  ros-kinetic-tf2-msgs ros-kinetic-tf2-py ros-kinetic-tf2-ros ros-kinetic-theora-image-transport
  ros-kinetic-topic-tools ros-kinetic-trajectory-msgs ros-kinetic-turtle-actionlib
  ros-kinetic-turtle-tf ros-kinetic-turtle-tf2 ros-kinetic-turtlesim ros-kinetic-urdf
  ros-kinetic-urdf-tutorial ros-kinetic-vision-opencv ros-kinetic-visualization-marker-tutorials
  ros-kinetic-visualization-msgs ros-kinetic-visualization-tutorials ros-kinetic-viz ros-kinetic-xacro
The following NEW packages will be installed:
  python3-catkin-pkg python3-catkin-pkg-modules
0 upgraded, 2 newly installed, 221 to remove and 3 not upgraded.
Need to get 0 B/44.8 kB of archives.
After this operation, 389 MB disk space will be freed.
Do you want to continue? [Y/n] n
Abort.
```

Is there any way to build this package? Should I build python3-catkin-pkg from source?

knorth55, 2019-09-15:
please try this command `sudo apt install python3-catkin-pkg-modules`
https://stackoverflow.com/questions/49221565/unable-to-use-cv-bridge-with-ros-kinetic-and-python3

sktometometo, 2019-09-15:
I have successfully build this package, thank you!
I have created a pull request about python3-<hoge>-modules dependencies. Please check [this](https://github.com/knorth55/coral_usb_ros/pull/4).

And I have got another error with sample launch.

```bash
sktometometo@Sherman ~ [gitbranch:] [workspace:/home/sktometometo/ros/ws_tpu_development] ->
 09:51 午後 日  9月 15$ roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=/image_publisher/output
... logging to /home/sktometometo/.ros/log/e2b568ee-d7b6-11e9-9339-507b9daced47/roslaunch-Sherman-16564.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Sherman:38553/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    edgetpu_object_detector (coral_usb/edgetpu_object_detector.py)

ROS_MASTER_URI=http://localhost:11311

process[edgetpu_object_detector-1]: started with pid [16582]
Traceback (most recent call last):
  File "/home/sktometometo/ros/ws_tpu_development/src/coral_usb_ros/node_scripts/edgetpu_object_detector.py", line 17, in <module>
    from chainercv.visualizations import vis_bbox
ImportError: No module named 'chainercv'
[edgetpu_object_detector-1] process has died [pid 16582, exit code 1, cmd /home/sktometometo/ros/ws_tpu_development/devel/lib/coral_usb/edgetpu_object_detector.py ~input:=/image_publisher/output __name:=edgetpu_object_detector __log:=/home/sktometometo/.ros/log/e2b568ee-d7b6-11e9-9339-507b9daced47/edgetpu_object_detector-1.log].
log file: /home/sktometometo/.ros/log/e2b568ee-d7b6-11e9-9339-507b9daced47/edgetpu_object_detector-1*.log
[edgetpu_object_detector-1] restarting process
process[edgetpu_object_detector-1]: started with pid [16588]
Traceback (most recent call last):
  File "/home/sktometometo/ros/ws_tpu_development/src/coral_usb_ros/node_scripts/edgetpu_object_detector.py", line 17, in <module>
    from chainercv.visualizations import vis_bbox
ImportError: No module named 'chainercv'
[edgetpu_object_detector-1] process has died [pid 16588, exit code 1, cmd /home/sktometometo/ros/ws_tpu_development/devel/lib/coral_usb/edgetpu_object_detector.py ~input:=/image_publisher/output __name:=edgetpu_object_detector __log:=/home/sktometometo/.ros/log/e2b568ee-d7b6-11e9-9339-507b9daced47/edgetpu_object_detector-1.log].
log file: /home/sktometometo/.ros/log/e2b568ee-d7b6-11e9-9339-507b9daced47/edgetpu_object_detector-1*.log
[edgetpu_object_detector-1] restarting process
process[edgetpu_object_detector-1]: started with pid [16594]
^C[edgetpu_object_detector-1] killing on exit
ImportError: numpy.core.multiarray failed to import
Traceback (most recent call last):
  File "/home/sktometometo/ros/ws_tpu_development/src/coral_usb_ros/node_scripts/edgetpu_object_detector.py", line 14, in <module>
    import cv2  # NOQA
ImportError: numpy.core.multiarray failed to import
shutting down processing monitor...
... shutting down processing monitor complete
done
```

This seemed because python3 environment did not have chainercv modules. and this error did not appear after instaling chainercv to python3 environment with pip3 command.
Is there any way to mark python3 library dependencies in catkin package.xml?

In addition, I have still got an error while executing the launch after chainercv installation to python3.

```bash
sktometometo@Sherman ~ [gitbranch:] [workspace:/home/sktometometo/ros/ws_tpu_development] ->
 09:54 午後 日  9月 15$ roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=/image_publisher/output
... logging to /home/sktometometo/.ros/log/e2b568ee-d7b6-11e9-9339-507b9daced47/roslaunch-Sherman-17864.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Sherman:41065/
[... 30 lines cut ...]
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/jsk_topic_tools/__init__.py", line 5, in <module>
    from .log_utils import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/jsk_topic_tools/log_utils.py", line 4, in <module>
    import cPickle as pickle
ImportError: No module named 'cPickle'
[edgetpu_object_detector-1] process has died [pid 17892, exit code 1, cmd /home/sktometometo/ros/ws_tpu_development/devel/lib/coral_usb/edgetpu_object_detector.py ~input:=/image_publisher/output __name:=edgetpu_object_detector __log:=/home/sktometometo/.ros/log/e2b568ee-d7b6-11e9-9339-507b9daced47/edgetpu_object_detector-1.log].
log file: /home/sktometometo/.ros/log/e2b568ee-d7b6-11e9-9339-507b9daced47/edgetpu_object_detector-1*.log
[edgetpu_object_detector-1] restarting process
process[edgetpu_object_detector-1]: started with pid [17904]
Traceback (most recent call last):
  File "/home/sktometometo/ros/ws_tpu_development/src/coral_usb_ros/node_scripts/edgetpu_object_detector.py", line 28, in <module>
    from jsk_topic_tools import ConnectionBasedTransport
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/jsk_topic_tools/__init__.py", line 5, in <module>
    from .log_utils import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/jsk_topic_tools/log_utils.py", line 4, in <module>
    import cPickle as pickle
ImportError: No module named 'cPickle'
^C[edgetpu_object_detector-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done**
```

This seems to be related to [this pull request](https://github.com/jsk-ros-pkg/jsk_common/pull/1636).
I am trying to build the version of jsk_topic_tools.

knorth55, 2019-09-15:
Did you do `source ~/ros/coral_ws/devel/setup.bash` before launching?

sktometometo, 2019-09-16:
I have succecfuly run the exapmle by building jsk_topic_tools and source the workspace.

closed 2019-09-16 by sktometometo