jsk-ros-pkg/coral_usb_ros #22: cv2 を見つけられない

shmpwk, 2020-05-28:
こんばんは．
[Run Edge TPU launch](https://github.com/knorth55/coral_usb_ros#run-edge-tpu-launch)でobject detectorを立ち上げようとしているのですが，`No module named 'cv2'`となっています．
恐らくパスが間違っているのかと思いますが，何か設定する必要があるのでしょうか．

knorth55, 2020-05-28:
```
source /opt/ros/melodic/setup.bash
source ~/coral_ws/devel/setup.bash
ipython
>> import cv2
```
これを試してエラーがでるかみてみてください

shmpwk, 2020-05-28:
エラー出ませんでした．

knorth55, 2020-05-28:
先のlaunchをするときにもう一度sourceをし直してやってみてください。
あとterminalの出力を全部はってもらえると助かります。

knorth55, 2020-05-28:
これはやりましたか？
`sudo apt-get install python3-opencv`

shmpwk, 2020-05-28:
すみません，opencvをアンインストールしていました．
インストールして無事動きました．お騒がせしました．

closed 2020-05-28 by shmpwk

knorth55, 2020-05-28:
あとの人のためにterminalの出力と何コマンドを実行したかをここに書いておいてください。

shmpwk, 2020-05-28:
はい，
[これ](https://github.com/jsk-ros-pkg/jsk_demos/issues/1272#issuecomment-542765613)を見て，セットアップしていました．
その手順におけるcoralの認識で
```
roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=/camera/rgb/image_raw
```
とした際に，terminal に `No module named 'cv2'`と出力されました．

参照した説明ではopencvのインストールについて書いてなかったので，
```
sudo apt-get install python3-opencv
```
をする必要がありました．(もともと自分でopencvを入れたつもりでいたが，入っていなかった．)

knorth55, 2020-05-28:
> roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=/camera/rgb/image_raw
とした際に，terminal に No module named 'cv2'と出力されました．

問題が発生したときには、問題が再現できるlaunchを共有する、他の環境で再現しない場合にはterminalの出力結果を全て共有する、ことが重要です。
なかなか忘れがちですが、できるだけ多くの情報を共有することが基本なので。
terminalの出力結果で、他のエラーを何かしら見逃している可能性があるので、issueやメールをする場合はterminalの**全ての出力結果**をコピーペーストしてくれると、デバグがしやすくてありがたいです。

shmpwk, 2020-05-29:
了解しました．以後気をつけます．

自分の環境ubuntu18.04, melodicでの
launchの出力全体は以下のようになっていました．
```
$ roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=/camera/rgb/image_raw
... logging to /home/shumpeiwakabayashi/.ros/log/4aaa27a2-a0e1-11ea-a66d-287fcf9948c3/roslaunch-shumpeiwakabayashi-ThinkPad-T495s-29977.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://shumpeiwakabayashi-ThinkPad-T495s:45251/

SUMMARY
========

PARAMETERS
 * /edgetpu_object_detector/label_file: /home/shumpeiwaka...
 * /edgetpu_object_detector/model_file: /home/shumpeiwaka...
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    edgetpu_object_detector (coral_usb/edgetpu_object_detector.py)

ROS_MASTER_URI=http://localhost:11311

process[edgetpu_object_detector-1]: started with pid [30985]
Traceback (most recent call last):
  File "/home/shumpeiwakabayashi/coral_ws/src/coral_usb_ros/node_scripts/edgetpu_object_detector.py", line 14, in <module>
    import cv2  # NOQA
ModuleNotFoundError: No module named 'cv2'
[edgetpu_object_detector-1] process has died [pid 30985, exit code 1, cmd /home/shumpeiwakabayashi/coral_ws/devel/lib/coral_usb/edgetpu_object_detector.py ~input:=/camera/rgb/image_raw __name:=edgetpu_object_detector __log:=/home/shumpeiwakabayashi/.ros/log/4aaa27a2-a0e1-11ea-a66d-287fcf9948c3/edgetpu_object_detector-1.log].
log file: /home/shumpeiwakabayashi/.ros/log/4aaa27a2-a0e1-11ea-a66d-287fcf9948c3/edgetpu_object_detector-1*.log
[edgetpu_object_detector-1] restarting process
```

knorth55, 2020-05-29:
ありがとうございます。