jsk-ros-pkg/coral_usb_ros #24: edgetpu_human_pose_estimatorを可視化する際のバグ

MiyabiTane, 2020-07-02:
北川さんが作って下さった部分のバグではないのですが、
```
roslaunch roseus_tutorials usb-camera.launch
roslaunch coral_usb edgetpu_human_pose_estimator.launch INPUT_IMAGE:=/image_raw
```
としてCoralを立ち上げた後、
```
rosrun image_view image_view image:=/edgetpu_human_pose_estimator/output/image
```
とするとエラーが出て可視化できなかったので、原因を調べたところ
https://gist.github.com/MiyabiTane/5282054021945c59c8202a0fdc307658#file-pose_engine-py-L99-L102
の部分でself.get_all_output_tensors_sizes()の中身は整数なのに、offsetが少数になってしまっていることが原因であることが分かりました。
そこで
https://gist.github.com/MiyabiTane/5282054021945c59c8202a0fdc307658#file-pose_engine-py-L103
のように直したところ、エラーがなくなり、結果を可視化することができました。

以前、@Kanazawanaoakiが自主プロでエラーなく動かすことができたという事例もあり、PCの環境に原因がある可能性もあるため、確認して頂きたく、issueにさせて頂きました、よろしくお願いします。@knorth55

knorth55, 2020-07-03:
これはエラーログはどんな感じですか？
エラーででている出力をここに全文貼ってもらえるとデバグに助かります。

knorth55, 2020-07-03:
このissueと同じですか？
https://github.com/google-coral/project-posenet/issues/27
https://github.com/google-coral/project-posenet/commit/f74ff7973e4b4349aaad9f50c7f0bc77fe33775b

MiyabiTane, 2020-07-04:
遅くなりましたすみません。

> このissueと同じですか？

同じです。

knorth55, 2020-07-05:
#25　で解決しました。

closed 2020-07-05 by knorth55