jsk-ros-pkg/coral_usb_ros #26: Problems with jsk_common / travis building docker image

hannes56a, 2020-07-08:
Hi,
first: Thanks a lot for this project.
I´m trying to build the docker image on a IOT2050 (ARM64) with a Coral TPU Chip (Mini PCIe Accelerator).
I changed some things in the Dockerfile
- changed "FROM ..." to "FROM ros_melodic", cause of ARM64
- added curl installation
- changed tensorflow-lite whl to ..."aarch64"
- changed tensorflow-lite whl to "cp37", because the ros:melodic brings python 3.6.9, and i updated it to 3.7.5

Dockerfile posted at the end.
I start building the docker image with

> sudo docker build --build-arg ROS_DISTRO=melodic --build-arg UBUNTU_VERSION=bionic -t knorth55/coral_usb_ros:melodic-latest .

Now I have an issue with jsk_common and/or travis in fc.rosinstall, after

> RUN cd ~ && \
>     mkdir -p ros/${ROS_DISTRO}/src && \
>     cd ros/${ROS_DISTRO}/src && \
>     wstool init && \
>     wstool set coral_usb_ros https://github.com/knorth55/coral_usb_ros.git -v master --git -y && \
>     wstool up -j 2 && \
>     wstool merge -y coral_usb_ros/fc.rosinstall && \
>     wstool merge -y coral_usb_ros/fc.rosinstall.${ROS_DISTRO} &&\
>     wstool up -j 2

... and I´m not able to solve it. Can please someone help me?
Here is the output error:

> [dynamic_reconfigure] Fetching https://github.com/ros/dynamic_reconfigure.git (version 1.6.3) to /root/ros/melodic/src/dynamic_reconfigure
> [dynamic_reconfigure] Done.
> [coral_usb_ros,dynamic_reconfigure] still active
> [coral_usb_ros,dynamic_reconfigure] still active
> Cloning into '/root/ros/melodic/src/jsk_common'...
> Cloning into '/root/ros/melodic/src/vision_opencv'...
> [vision_opencv] Fetching https://github.com/ros-perception/vision_opencv.git (version melodic) to /root/ros/melodic/src/vision_opencv
> [vision_opencv] Done.
> [jsk_common,vision_opencv] still active
> [jsk_common,vision_opencv] still active
> [jsk_common] still active
> [jsk_common] still active
> [...]
> [jsk_common] still active
> [jsk_common] still active
> [jsk_common]WARNING [vcstools] Command failed: 'git submodule update --init --recursive'
>  run at: '/root/ros/melodic/src/jsk_common'
>  errcode: 1:
> Submodule '.travis' (git://github.com/jsk-ros-pkg/jsk_travis) registered for path '.travis'
> Cloning into '/root/ros/melodic/src/jsk_common/.travis'...
> fatal: unable to connect to github.com:
> github.com[0: 140.82.118.3]: errno=Connection timed out
> 
> fatal: clone of 'git://github.com/jsk-ros-pkg/jsk_travis' into submodule path '/root/ros/melodic/src/jsk_common/.travis' failed
> Failed to clone '.travis'. Retry scheduled
> Cloning into '/root/ros/melodic/src/jsk_common/.travis'...
> fatal: unable to connect to github.com:
> github.com[0: 140.82.118.3]: errno=Connection timed out
> 
> fatal: clone of 'git://github.com/jsk-ros-pkg/jsk_travis' into submodule path '/root/ros/melodic/src/jsk_common/.travis' failed
> Failed to clone '.travis' a second time, aborting
> [/vcstools]
> [jsk_common] Fetching https://github.com/jsk-ros-pkg/jsk_common.git (version master) to /root/ros/melodic/src/jsk_common
>  still active
> [jsk_common] still active
> [jsk_common] still active
> [...]
> [jsk_common] still active
> [jsk_common] still active
> Exception caught during install: Error processing 'jsk_common' : [jsk_common] Checkout of https://github.com/jsk-ros-pkg/jsk_common.git version master into /root/ros/melodic/src/jsk_common failed.
> 
> ERROR in config: Error processing 'jsk_common' : [jsk_common] Checkout of https://github.com/jsk-ros-pkg/jsk_common.git version master into /root/ros/melodic/src/jsk_common failed.
> 
> The command '/bin/sh -c cd ~ &&     mkdir -p ros/${ROS_DISTRO}/src &&     cd ros/${ROS_DISTRO}/src &&     wstool init &&     wstool set coral_usb_ros https://github.com/knorth55/coral_usb_ros.git -v master --git -y &&     wstool up -j 2 &&     wstool merge -y coral_usb_ros/fc.rosinstall &&     wstool merge -y coral_usb_ros/fc.rosinstall.${ROS_DISTRO} &&    wstool up -j 2' returned a non-zero code: 1

Dockerfile:

> ARG ROS_DISTRO
> ARG UBUNTU_VERSION
> FROM ros:melodic
> ENV DEBIAN_FRONTEND noninteractive
> 
> RUN python3 --version
> RUN apt-get update -y && \
>     apt-get install -y python3.7
> 
> RUN rm /usr/bin/python3
> RUN sudo ln -s python3.7 /usr/bin/python3
> 
> RUN python3 --version
> 
> RUN echo "deb http://packages.ros.org/ros-testing/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-la$
> RUN apt-get update && apt-get install -y wget git
> RUN wget http://packages.ros.org/ros.key -O - | apt-key add -
> 
> #### FIXME: https://github.com/start-jsk/jsk_apc/pull/2664
> RUN apt-get update && apt-get dist-upgrade -y && apt-get install -y \
>     apt-utils \
>     python-catkin-tools \
>     python-rosdep \
>     python-setuptools \
>     python-wstool \
>     python-pip
> 
> RUN pip install 'pip==9.0.3' && \
>     pip install 'setuptools==44.1.0'
> 
> RUN apt-get update && \
>     apt-get install curl
> 
> RUN apt-get update && \
>     apt-get install -y apt-transport-https ca-certificates && \
>     echo "deb https://packages.cloud.google.com/apt coral-edgetpu-stable main" | tee /etc/apt/sources.list.d/coral$
>     curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | apt-key add - && \
>     apt-get update && \
>     apt-get install -y libedgetpu1-std python3-edgetpu python3-pip
> 
> ARG ROS_DISTRO
> RUN if [ ${ROS_DISTRO} = "kinetic" ]; then \
>       wget https://dl.google.com/coral/python/tflite_runtime-1.14.0-cp35-cp35m-linux_aarch64.whl && \
>       pip3 install tflite_runtime-1.14.0-cp35-cp35m-linux_aarch64.whl; \
>     else \
>       wget https://dl.google.com/coral/python/tflite_runtime-1.14.0-cp37-cp37m-linux_aarch64.whl && \
>       pip3 install tflite_runtime-1.14.0-cp37-cp37m-linux_aarch64.whl; \
>     fi
> 
> ARG ROS_DISTRO
> RUN cd ~ && \
>     mkdir -p ros/${ROS_DISTRO}/src && \
>     cd ros/${ROS_DISTRO}/src && \
>     wstool init && \
>     wstool set coral_usb_ros https://github.com/knorth55/coral_usb_ros.git -v master --git -y && \
>     wstool up -j 2 && \
>     wstool merge -y coral_usb_ros/fc.rosinstall && \
>     wstool merge -y coral_usb_ros/fc.rosinstall.${ROS_DISTRO} &&\
>     wstool up -j 2
> 
> RUN rosdep update --include-eol-distros
> 
> #### /opt/ros/${ROS_DISTRO}/share can be changed after rosdep install, so we run it 3 times.
> RUN apt-get update && \
>     for i in $(seq 3); do \
>       rosdep install --rosdistro ${ROS_DISTRO} -r -y -i --from-paths /opt/ros/${ROS_DISTRO}/share ~/ros/${ROS_DIST$
>     done
>     
> ARG ROS_DISTRO
> RUN . /opt/ros/${ROS_DISTRO}/setup.sh && \
>     cd ~/ros/${ROS_DISTRO} && \
>     if [ ${ROS_DISTRO} = "kinetic" ] ; then \
>       catkin config -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.5m -DPYTHON_LIB$
>     else \
>       catkin config -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIB$
>     fi && \
>     catkin build

knorth55, 2020-07-09:
@hannes56a It seems that your docker image do not have internet access.
can you check if you can do `ping 8.8.8.8` inside your docker image?
also, can you check if you can do `git clone https:/github.com/jsk-ros-pkg/jsk_common.git` inside your docker image?

knorth55, 2020-07-09:
@hannes56a BTW, `19.03` is not supported for `ROS Melodic` so it is better to use `18.04`.

hannes56a, 2020-07-13:
@ knorth55 Thank you for your reply. But internet access is ok. Sorry for the late answer, im currently in holiday ;)

knorth55, 2020-07-13:
@hannes56a no problem. 
your error log says you cannot clone repository by `https`.
you should check if you can clone in your environment.

hannes56a, 2020-07-16:
Hi, now it works (not the hole dockerfile but this issue). Im not sure why. I did two things:
1. Using my home internet connection instead of the in the office
2. Install openssh befor cloning from https: apt-get install ssh-client

Now i have another issue, but i think its an "out of memory" issue. I have to try using swap file...

knorth55, 2020-07-16:
Good, i will close this issue.
if u have another problem, please open another issue.

closed 2020-07-16 by knorth55