jsk-ros-pkg/coral_usb_ros #28: When I execute `catkin build`, the error was occurred at the first time.

mqcmd196, 2020-09-04:
I executed some commands and `catkin build` as your readme shows, it returned the error at the first time.
```
Errors     << coral_usb:make /home/tork/coral_ws/logs/coral_usb/build.make.000.log
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] 許可がありません: '/home/tork/.cache/pip/wheels/97/cc/70/f8b0783d465997368ed7b3a1196b9cde33c1eb32fdb3f2580a'
Consider using the `--user` option or check the permissions.

ERROR: Could not install packages due to an EnvironmentError: [Errno 13] 許可がありません: '/home/tork/.cache/pip/wheels/97/cc/70/f8b0783d465997368ed7b3a1196b9cde33c1eb32fdb3f2580a'
Consider using the `--user` option or check the permissions.

Error, clearing virtualenv and retrying: Command '['/home/tork/coral_ws/build/coral_usb/venv/bin/python', '-m', 'pip', 'install', '-qq', '-r', '/home/tork/coral_ws/build/coral_usb/generated_requirements.txt']' returned non-zero exit status 1
Error, clearing virtualenv and retrying: Command '['/home/tork/coral_ws/devel/.private/coral_usb/share/coral_usb/venv/bin/python', '-m', 'pip', 'install', '-qq', '-r', '/home/tork/coral_ws/build/coral_usb/generated_requirements.txt']' returned non-zero exit status 1
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] 許可がありません: '/home/tork/.cache/pip/wheels/97/cc/70/f8b0783d465997368ed7b3a1196b9cde33c1eb32fdb3f2580a'
Consider using the `--user` option or check the permissions.

Error, clearing virtualenv and retrying: Command '['/home/tork/coral_ws/build/coral_usb/venv/bin/python', '-m', 'pip', 'install', '-qq', '-r', '/home/tork/coral_ws/build/coral_usb/generated_requirements.txt']' returned non-zero exit status 1
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] 許可がありません: '/home/tork/.cache/pip/wheels/97/cc/70/f8b0783d465997368ed7b3a1196b9cde33c1eb32fdb3f2580a'
Consider using the `--user` option or check the permissions.

Error, clearing virtualenv and retrying: Command '['/home/tork/coral_ws/devel/.private/coral_usb/share/coral_usb/venv/bin/python', '-m', 'pip', 'install', '-qq', '-r', '/home/tork/coral_ws/build/coral_usb/generated_requirements.txt']' returned non-zero exit status 1
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] 許可がありません: '/home/tork/.cache/pip/wheels/97/cc/70/f8b0783d465997368ed7b3a1196b9cde33c1eb32fdb3f2580a'
Consider using the `--user` option or check the permissions.

Error, clearing virtualenv and retrying: Command '['/home/tork/coral_ws/build/coral_usb/venv/bin/python', '-m', 'pip', 'install', '-qq', '-r', '/home/tork/coral_ws/build/coral_usb/generated_requirements.txt']' returned non-zero exit status 1
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] 許可がありません: '/home/tork/.cache/pip/wheels/97/cc/70/f8b0783d465997368ed7b3a1196b9cde33c1eb32fdb3f2580a'
Consider using the `--user` option or check the permissions.

Error, clearing virtualenv and retrying: Command '['/home/tork/coral_ws/devel/.private/coral_usb/share/coral_usb/venv/bin/python', '-m', 'pip', 'install', '-qq', '-r', '/home/tork/coral_ws/build/coral_usb/generated_requirements.txt']' returned non-zero exit status 1
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] 許可がありません: '/home/tork/.cache/pip/wheels/97/cc/70/f8b0783d465997368ed7b3a1196b9cde33c1eb32fdb3f2580a'
Consider using the `--user` option or check the permissions.

Traceback (most recent call last):
  File "/opt/ros/melodic/lib/catkin_virtualenv/build_venv", line 101, in <module>
    deploy.install_dependencies()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/dh_virtualenv/deployment.py", line 191, in install_dependencies
    check_call(self.pip('-r', requirements_path))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/dh_virtualenv/deployment.py", line 39, in check_call
    return subprocess.check_call(cmd, *args, **kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/tork/coral_ws/build/coral_usb/venv/bin/python', '-m', 'pip', 'install', '-qq', '-r', '/home/tork/coral_ws/build/coral_usb/generated_requirements.txt']' returned non-zero exit status 1
make[2]: *** [venv] Error 1
make[2]: *** 未完了のジョブを待っています....
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] 許可がありません: '/home/tork/.cache/pip/wheels/97/cc/70/f8b0783d465997368ed7b3a1196b9cde33c1eb32fdb3f2580a'
Consider using the `--user` option or check the permissions.

Traceback (most recent call last):
  File "/opt/ros/melodic/lib/catkin_virtualenv/build_venv", line 101, in <module>
    deploy.install_dependencies()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/dh_virtualenv/deployment.py", line 191, in install_dependencies
    check_call(self.pip('-r', requirements_path))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/dh_virtualenv/deployment.py", line 39, in check_call
    return subprocess.check_call(cmd, *args, **kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/tork/coral_ws/devel/.private/coral_usb/share/coral_usb/venv/bin/python', '-m', 'pip', 'install', '-qq', '-r', '/home/tork/coral_ws/build/coral_usb/generated_requirements.txt']' returned non-zero exit status 1
make[2]: *** [/home/tork/coral_ws/devel/.private/coral_usb/share/coral_usb/venv] Error 1
make[1]: *** [CMakeFiles/coral_usb_generate_virtualenv.dir/all] Error 2
make: *** [all] Error 2
cd /home/tork/coral_ws/build/coral_usb; catkin build --get-env coral_usb | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << coral_usb:make                         [ Exited with code 2 ]    
Failed    <<< coral_usb                              [ 40.9 seconds ]          
[build] Summary: 14 of 15 packages succeeded.                                  
[build]   Ignored:   2 packages were skipped or are blacklisted.               
[build]   Warnings:  2 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 2 minutes and 2.3 seconds total.                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```
[... 41 lines cut ...]
Finished  <<< image_geometry                         [ 0.3 seconds ]          
Starting  >>> image_view2                                                     
Finished  <<< jsk_tilt_laser                         [ 0.2 seconds ]          
Finished  <<< dynamic_tf_publisher                   [ 0.4 seconds ]          
Finished  <<< opencv_tests                           [ 0.2 seconds ]          
Starting  >>> jsk_topic_tools                                                 
Finished  <<< jsk_network_tools                      [ 0.6 seconds ]          
Finished  <<< image_view2                            [ 0.4 seconds ]          
Finished  <<< jsk_topic_tools                        [ 0.4 seconds ]          
Starting  >>> jsk_data                                                        
Starting  >>> jsk_tools                                                       
Starting  >>> coral_usb                                                       
Finished  <<< jsk_tools                              [ 0.2 seconds ]          
Starting  >>> multi_map_server                                                
Finished  <<< coral_usb                              [ 0.2 seconds ]          
Finished  <<< jsk_data                               [ 0.2 seconds ]          
Finished  <<< multi_map_server                       [ 0.2 seconds ]          
[build] Summary: All 14 packages succeeded!                                   
[build]   Ignored:   2 packages were skipped or are blacklisted.              
[build]   Warnings:  None.                                                    
[build]   Abandoned: None.                                                    
[build]   Failed:    None.                                                    
[build] Runtime: 2.0 seconds total.
```
and it finished with no error. I don't know how important this error is, so I reported here just in case.

knorth55, 2020-09-04:
Yes, the first error is pretty important.
You need to complete it, otherwise this package does not work.

```bash
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] 許可がありません: '/home/tork/.cache/pip/wheels/97/cc/70/f8b0783d465997368ed7b3a1196b9cde33c1eb32fdb3f2580a'
Consider using the `--user` option or check the permissions.
```
from above error, it seems you have wrong permission for `/home/tork/.cache`
can you check the permission of the directory like ` ls -alF /home/tork` or `mkdir /home/tork/.cache/test -p`?

mqcmd196, 2020-09-04:
I executed 2 commands.
```bash
$ ls -alF /home/tork/ | grep .cache
drwx------  4 root root  4096  8月 24 14:27 .cache/
```
```bash
$ mkdir /home/tork/.cache/test -p
mkdir: ディレクトリ `/home/tork/.cache' を作成できません: 許可がありません
```
Should I modify the permission?

knorth55, 2020-09-04:
yes, you should.
you should change the ownership and mod of the directory.
maybe you made the directory with sudo command.

```
sudo chown tork:tork /home/tork/.cache
```

mqcmd196, 2020-09-05:
This problem had been solved. #29

closed 2020-09-05 by mqcmd196