jsk-ros-pkg/coral_usb_ros #29: When I launch edgetpu_human_pose_estimator.launch, the python script returns SyntaxError.

mqcmd196, 2020-09-04:
I launched `edgetpu_human_pose_estimator.launch`, and terminal showed this error.
```
[edgetpu_human_pose_estimator-2] restarting process
process[edgetpu_human_pose_estimator-2]: started with pid [15649]
Traceback (most recent call last):
  File "/home/tork/coral_ws/src/coral_usb_ros/node_scripts/edgetpu_human_pose_estimator.py", line 16, in <module>
    from chainercv.visualizations import vis_point
  File "/usr/local/lib/python2.7/dist-packages/chainercv/__init__.py", line 3, in <module>
    from chainercv import chainer_experimental  # NOQA
  File "/usr/local/lib/python2.7/dist-packages/chainercv/chainer_experimental/__init__.py", line 1, in <module>
    from chainercv.chainer_experimental import datasets  # NOQA
  File "/usr/local/lib/python2.7/dist-packages/chainercv/chainer_experimental/datasets/__init__.py", line 1, in <module>
    from chainercv.chainer_experimental.datasets import sliceable  # NOQA
  File "/usr/local/lib/python2.7/dist-packages/chainercv/chainer_experimental/datasets/sliceable/__init__.py", line 1, in <module>
    from chainercv.chainer_experimental.datasets.sliceable.sliceable_dataset import SliceableDataset  # NOQA
  File "/usr/local/lib/python2.7/dist-packages/chainercv/chainer_experimental/datasets/sliceable/sliceable_dataset.py", line 5, in <module>
    import chainer
  File "/usr/local/lib/python2.7/dist-packages/chainer/__init__.py", line 10, in <module>
    from chainer import backends  # NOQA
  File "/usr/local/lib/python2.7/dist-packages/chainer/backends/__init__.py", line 1, in <module>
    from chainer.backends import cuda  # NOQA
  File "/usr/local/lib/python2.7/dist-packages/chainer/backends/cuda.py", line 77
    def shape(self) -> types.Shape:
                    ^
SyntaxError: invalid syntax
[edgetpu_human_pose_estimator-2] process has died [pid 15649, exit code 1, cmd /home/tork/coral_ws/devel/lib/coral_usb/edgetpu_human_pose_estimator.py ~input:=/image_publisher/output __name:=edgetpu_human_pose_estimator __log:=/home/tork/.ros/log/74102cae-ee6b-11ea-8bc9-c0b883e85b16/edgetpu_human_pose_estimator-2.log].
log file: /home/tork/.ros/log/74102cae-ee6b-11ea-8bc9-c0b883e85b16/edgetpu_human_pose_estimator-2*.log
```

I thought this script should be run on python3 environment, so I confirmed my environment by `catkin config`.
```
-------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/tork/coral_ws
-------------------------------------------------------------------------------
Build Space:        [exists] /home/tork/coral_ws/build
Devel Space:        [exists] /home/tork/coral_ws/devel
Install Space:      [unused] /home/tork/coral_ws/install
Log Space:          [exists] /home/tork/coral_ws/logs
Source Space:       [exists] /home/tork/coral_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------------------------
Additional CMake Args:       -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------------------------
```
Please tell me how to fix this problem. Sorry for submitting 2 issues at the same time. I also show you the result of `pip freeze | grep chainer`.  `pip3 freeze | grep chainer` returns nothing.
```
chainer==7.0.0
chainercv==0.13.1
```

knorth55, 2020-09-04:
it seems you don't install package correctly because all the programs are running on python2.7.
all the programs on this package should run on python3.
did you source coral tpu workspace as `source ~/coral_ws/devel/setup.bash` before launching it?

mqcmd196, 2020-09-04:
Yes, I sourced `coral/ws/devel/setup.bash` certainly.
```
tork@oem-ThinkPad-T490 ~ 
$ source /opt/ros/melodic/setup.bash
tork@oem-ThinkPad-T490 ~ 
$ source ~/coral_ws/devel/setup.bash
tork@oem-ThinkPad-T490 ~ 
$ roslaunch coral_usb edgetpu_human_pose_estimator.launch INPUT_IMAGE:=/image_publisher/output
... logging to /home/tork/.ros/log/009722c2-ee80-11ea-8bc9-c0b883e85b16/roslaunch-oem-ThinkPad-T490-25512.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://oem-ThinkPad-T490:33703/

SUMMARY
========

PARAMETERS
 * /edgetpu_human_pose_estimator/model_file: /home/tork/coral_...
 * /rosdistro: melodic
 * /rosversion: 1.14.9

NODES
  /
    edgetpu_human_pose_estimator (coral_usb/edgetpu_human_pose_estimator.py)

auto-starting new master
process[master]: started with pid [25522]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 009722c2-ee80-11ea-8bc9-c0b883e85b16
process[rosout-1]: started with pid [25533]
started core service [/rosout]
process[edgetpu_human_pose_estimator-2]: started with pid [25540]
Traceback (most recent call last):
  File "/home/tork/coral_ws/src/coral_usb_ros/node_scripts/edgetpu_human_pose_estimator.py", line 16, in <module>
    from chainercv.visualizations import vis_point
  File "/usr/local/lib/python2.7/dist-packages/chainercv/__init__.py", line 3, in <module>
    from chainercv import chainer_experimental  # NOQA
  File "/usr/local/lib/python2.7/dist-packages/chainercv/chainer_experimental/__init__.py", line 1, in <module>
    from chainercv.chainer_experimental import datasets  # NOQA
  File "/usr/local/lib/python2.7/dist-packages/chainercv/chainer_experimental/datasets/__init__.py", line 1, in <module>
    from chainercv.chainer_experimental.datasets import sliceable  # NOQA
  File "/usr/local/lib/python2.7/dist-packages/chainercv/chainer_experimental/datasets/sliceable/__init__.py", line 1, in <module>
    from chainercv.chainer_experimental.datasets.sliceable.sliceable_dataset import SliceableDataset  # NOQA
  File "/usr/local/lib/python2.7/dist-packages/chainercv/chainer_experimental/datasets/sliceable/sliceable_dataset.py", line 5, in <module>
    import chainer
  File "/usr/local/lib/python2.7/dist-packages/chainer/__init__.py", line 10, in <module>
    from chainer import backends  # NOQA
  File "/usr/local/lib/python2.7/dist-packages/chainer/backends/__init__.py", line 1, in <module>
    from chainer.backends import cuda  # NOQA
  File "/usr/local/lib/python2.7/dist-packages/chainer/backends/cuda.py", line 77
    def shape(self) -> types.Shape:
                    ^
SyntaxError: invalid syntax
[edgetpu_human_pose_estimator-2] process has died [pid 25540, exit code 1, cmd /home/tork/coral_ws/devel/lib/coral_usb/edgetpu_human_pose_estimator.py ~input:=/image_publisher/output __name:=edgetpu_human_pose_estimator __log:=/home/tork/.ros/log/009722c2-ee80-11ea-8bc9-c0b883e85b16/edgetpu_human_pose_estimator-2.log].
log file: /home/tork/.ros/log/009722c2-ee80-11ea-8bc9-c0b883e85b16/edgetpu_human_pose_estimator-2*.log
```

knorth55, 2020-09-04:
maybe we need to solve #28 first to build correctly.

ShreyasSkandanS, 2020-09-04:
I've got the same error but didn't experience any of the issues in #28.

knorth55, 2020-09-04:
@ShreyasSkandanS @mqcmd196 
can you clean up all the workspace, update `coral_usb_ros` to master and build workspace again?
also can you paste the output of the first `catkin build` ?

knorth55, 2020-09-04:
@ShreyasSkandanS @mqcmd196 
can you also let me know the version of `catkin_virtualenv`?
we use `0.6.1` for latest build.

ShreyasSkandanS, 2020-09-04:
I ran catkin build again on a fresh workspace, and it seems to now break here:
`Errors     << coral_usb:make /home/shreyas/coral_ws/logs/coral_usb/build.make.000.log   
ERROR: Could not find a version that satisfies the requirement numpy==1.18.5 (from -r /home/shreyas/coral_ws/build/coral_usb/generated_requirements.txt (line 4)) (from versions: 1.3.0, 1.4.1, 1.5.0, 1.5.1, 1.6.0, 1.6.1, 1.6.2, 1.7.0, 1.7.1, 1.7.2, 1.8.0, 1.8.1, 1.8.2, 1.9.0, 1.9.1, 1.9.2, 1.9.3, 1.10.0, 1.10.0.post2, 1.10.1, 1.10.2, 1.10.3, 1.10.4, 1.11.0b3, 1.11.0rc1, 1.11.0rc2, 1.11.0, 1.11.1rc1, 1.11.1, 1.11.2rc1, 1.11.2, 1.11.3, 1.12.0b1, 1.12.0rc1, 1.12.0rc2, 1.12.0, 1.12.1rc1, 1.12.1, 1.13.0rc1, 1.13.0rc2, 1.13.0, 1.13.1, 1.13.3, 1.14.0rc1, 1.14.0, 1.14.1, 1.14.2, 1.14.3, 1.14.4, 1.14.5, 1.14.6, 1.15.0rc1, 1.15.0rc2, 1.15.0, 1.15.1, 1.15.2, 1.15.3, 1.15.4, 1.16.0rc1, 1.16.0rc2, 1.16.0, 1.16.1, 1.16.2, 1.16.3, 1.16.4, 1.16.5, 1.16.6)
ERROR: No matching distribution found for numpy==1.18.5 (from -r /home/shreyas/coral_ws/build/coral_usb/generated_requirements.txt (line 4))
ERROR: Could not find a version that satisfies the requirement numpy==1.18.5 (from -r /home/shreyas/coral_ws/build/coral_usb/generated_requirements.txt (line 4)) (from versions: 1.3.0, 1.4.1, 1.5.0, 1.5.1, 1.6.0, 1.6.1, 1.6.2, 1.7.0, 1.7.1, 1.7.2, 1.8.0, 1.8.1, 1.8.2, 1.9.0, 1.9.1, 1.9.2, 1.9.3, 1.10.0, 1.10.0.post2, 1.10.1, 1.10.2, 1.10.3, 1.10.4, 1.11.0b3, 1.11.0rc1, 1.11.0rc2, 1.11.0, 1.11.1rc1, 1.11.1, 1.11.2rc1, 1.11.2, 1.11.3, 1.12.0b1, 1.12.0rc1, 1.12.0rc2, 1.12.0, 1.12.1rc1, 1.12.1, 1.13.0rc1, 1.13.0rc2, 1.13.0, 1.13.1, 1.13.3, 1.14.0rc1, 1.14.0, 1.14.1, 1.14.2, 1.14.3, 1.14.4, 1.14.5, 1.14.6, 1.15.0rc1, 1.15.0rc2, 1.15.0, 1.15.1, 1.15.2, 1.15.3, 1.15.4, 1.16.0rc1, 1.16.0rc2, 1.16.0, 1.16.1, 1.16.2, 1.16.3, 1.16.4, 1.16.5, 1.16.6)
ERROR: No matching distribution found for numpy==1.18.5 (from -r /home/shreyas/coral_ws/build/coral_usb/generated_requirements.txt (line 4))
Error, clearing virtualenv and retrying: Command '['/home/shreyas/coral_ws/build/coral_usb/venv/bin/python', '-m', 'pip', 'install', '-qq', '-r', '/home/shreyas/coral_ws/build/coral_usb/generated_requirements.txt']' returned non-zero exit status 1
Error, clearing virtualenv and retrying: Command '['/home/shreyas/coral_ws/devel/.private/coral_usb/share/coral_usb/venv/bin/python', '-m', 'pip', 'install', '-qq', '-r', '/home/shreyas/coral_ws/build/coral_usb/generated_requirements.txt']' returned non-zero exit status 1
ERROR: Could not find a version that satisfies the requirement numpy==1.18.5 (from -r /home/shreyas/coral_ws/build/coral_usb/generated_requirements.txt (line 4)) (from versions: 1.3.0, 1.4.1, 1.5.0, 1.5.1, 1.6.0, 1.6.1, 1.6.2, 1.7.0, 1.7.1, 1.7.2, 1.8.0, 1.8.1, 1.8.2, 1.9.0, 1.9.1, 1.9.2, 1.9.3, 1.10.0, 1.10.0.post2, 1.10.1, 1.10.2, 1.10.3, 1.10.4, 1.11.0b3, 1.11.0rc1, 1.11.0rc2, 1.11.0, 1.11.1rc1, 1.11.1, 1.11.2rc1, 1.11.2, 1.11.3, 1.12.0b1, 1.12.0rc1, 1.12.0rc2, 1.12.0, 1.12.1rc1, 1.12.1, 1.13.0rc1, 1.13.0rc2, 1.13.0, 1.13.1, 1.13.3, 1.14.0rc1, 1.14.0, 1.14.1, 1.14.2, 1.14.3, 1.14.4, 1.14.5, 1.14.6, 1.15.0rc1, 1.15.0rc2, 1.15.0, 1.15.1, 1.15.2, 1.15.3, 1.15.4, 1.16.0rc1, 1.16.0rc2, 1.16.0, 1.16.1, 1.16.2, 1.16.3, 1.16.4, 1.16.5, 1.16.6)
ERROR: No matching distribution found for numpy==1.18.5 (from -r /home/shreyas/coral_ws/build/coral_usb/generated_requirements.txt (line 4))
ERROR: Could not find a version that satisfies the requirement numpy==1.18.5 (from -r /home/shreyas/coral_ws/build/coral_usb/generated_requirements.txt (line 4)) (from versions: 1.3.0, 1.4.1, 1.5.0, 1.5.1, 1.6.0, 1.6.1, 1.6.2, 1.7.0, 1.7.1, 1.7.2, 1.8.0, 1.8.1, 1.8.2, 1.9.0, 1.9.1, 1.9.2, 1.9.3, 1.10.0, 1.10.0.post2, 1.10.1, 1.10.2, 1.10.3, 1.10.4, 1.11.0b3, 1.11.0rc1, 1.11.0rc2, 1.11.0, 1.11.1rc1, 1.11.1, 1.11.2rc1, 1.11.2, 1.11.3, 1.12.0b1, 1.12.0rc1, 1.12.0rc2, 1.12.0, 1.12.1rc1, 1.12.1, 1.13.0rc1, 1.13.0rc2, 1.13.0, 1.13.1, 1.13.3, 1.14.0rc1, 1.14.0, 1.14.1, 1.14.2, 1.14.3, 1.14.4, 1.14.5, 1.14.6, 1.15.0rc1, 1.15.0rc2, 1.15.0, 1.15.1, 1.15.2, 1.15.3, 1.15.4, 1.16.0rc1, 1.16.0rc2, 1.16.0, 1.16.1, 1.16.2, 1.16.3, 1.16.4, 1.16.5, 1.16.6)
ERROR: No matching distribution found for numpy==1.18.5 (from -r /home/shreyas/coral_ws/build/coral_usb/generated_requirements.txt (line 4))
Error, clearing virtualenv and retrying: Command '['/home/shreyas/coral_ws/build/coral_usb/venv/bin/python', '-m', 'pip', 'install', '-qq', '-r', '/home/shreyas/coral_ws/build/coral_usb/generated_requirements.txt']' returned non-zero exit status 1
Error, clearing virtualenv and retrying: Command '['/home/shreyas/coral_ws/devel/.private/coral_usb/share/coral_usb/venv/bin/python', '-m', 'pip', 'install', '-qq', '-r', '/home/shreyas/coral_ws/build/coral_usb/generated_requirements.txt']' returned non-zero exit status 1
ERROR: Could not find a version that satisfies the requirement numpy==1.18.5 (from -r /home/shreyas/coral_ws/build/coral_usb/generated_requirements.txt (line 4)) (from versions: 1.3.0, 1.4.1, 1.5.0, 1.5.1, 1.6.0, 1.6.1, 1.6.2, 1.7.0, 1.7.1, 1.7.2, 1.8.0, 1.8.1, 1.8.2, 1.9.0, 1.9.1, 1.9.2, 1.9.3, 1.10.0, 1.10.0.post2, 1.10.1, 1.10.2, 1.10.3, 1.10.4, 1.11.0b3, 1.11.0rc1, 1.11.0rc2, 1.11.0, 1.11.1rc1, 1.11.1, 1.11.2rc1, 1.11.2, 1.11.3, 1.12.0b1, 1.12.0rc1, 1.12.0rc2, 1.12.0, 1.12.1rc1, 1.12.1, 1.13.0rc1, 1.13.0rc2, 1.13.0, 1.13.1, 1.13.3, 1.14.0rc1, 1.14.0, 1.14.1, 1.14.2, 1.14.3, 1.14.4, 1.14.5, 1.14.6, 1.15.0rc1, 1.15.0rc2, 1.15.0, 1.15.1, 1.15.2, 1.15.3, 1.15.4, 1.16.0rc1, 1.16.0rc2, 1.16.0, 1.16.1, 1.16.2, 1.16.3, 1.16 [... cut ...]
Additional CMake Args:       -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------------------------------------
Workspace configuration appears valid.
`

knorth55, 2020-09-04:
I follow the readme document, build all the workspace on my melodic computer and it works.
[image]

knorth55, 2020-09-04:
@ShreyasSkandanS it seems you are using old `catkin_virtualenv`.
can you get the version information?
```bash
$ rosversion catkin_virtualenv
0.6.1
```

ShreyasSkandanS, 2020-09-04:
I'm using 0.5.2.

ShreyasSkandanS, 2020-09-04:
> @ShreyasSkandanS it seems you are using old `catkin_virtualenv`.
> can you get the version information?
> 
> ```shell
> $ rosversion catkin_virtualenv
> 0.6.1
> ```

What's the best way to upgrade to 0.6.1?

knorth55, 2020-09-04:
@ShreyasSkandanS 

how about just running `sudo apt install ros-melodic-catkin-virtualenv`?
which version can you install with that command?
if you get lower version, you can build it from source or use TestingRepository `ros-testing` for installation
http://wiki.ros.org/TestingRepository

knorth55, 2020-09-04:
I found out that normal ros repository release `0.5.2` for `catkin_virtualenv` and testing repository release `0.6.1`.
http://packages.ros.org/ros/ubuntu/pool/main/r/ros-melodic-catkin-virtualenv/
http://packages.ros.org/ros-testing/ubuntu/pool/main/r/ros-melodic-catkin-virtualenv/

ShreyasSkandanS, 2020-09-04:
> I found out that normal ros repository release `0.5.2` for `catkin_virtualenv` and testing repository release `0.6.1`.
> http://packages.ros.org/ros/ubuntu/pool/main/r/ros-melodic-catkin-virtualenv/
> http://packages.ros.org/ros-testing/ubuntu/pool/main/r/ros-melodic-catkin-virtualenv/

That worked! Thank you very much for the help.

I also had to make a small change to the **urllib** functions in **download_models.py**

`
#!/usr/bin/env python

import os
import tarfile
import urllib.request

import rospkg

url = 'https://dl.google.com/coral/canned_models/all_models.tar.gz'
seg_urls = [
    'https://github.com/google-coral/edgetpu/raw/master/test_data/'
    'deeplabv3_mnv2_dm05_pascal_quant_edgetpu.tflite',
    'https://github.com/google-coral/edgetpu/raw/master/test_data/'
    'deeplabv3_mnv2_pascal_quant_edgetpu.tflite',
]

rospack = rospkg.RosPack()
pkg_path = rospack.get_path('coral_usb')
models_path = os.path.join(pkg_path, './models')
tar_path = os.path.join(models_path, 'all_models.tar.gz')
if not os.path.exists(models_path):
    os.makedirs(models_path)
urllib.request.urlretrieve(url, tar_path)

with tarfile.open(tar_path) as tar_f:
    tar_f.extractall(models_path)

for seg_url in seg_urls:
    filename = seg_url.split('/')[-1]
    filepath = os.path.join(models_path, filename)
    urllib.request.urlretrieve(seg_url, filepath)

`

knorth55, 2020-09-04:
@ShreyasSkandanS thank you very much for the bug in `download_models.py`
I made a bug fix for that.
https://github.com/knorth55/coral_usb_ros/commit/f0e6ac86cec558e54bf127cce164f082b94d5949

Now you can run it with no error in latest version.

knorth55, 2020-09-05:
@mqcmd196 anyway, did you solve your problem?

mqcmd196, 2020-09-05:
@ShreyasSkandanS @knorth55 Thanks a lot! 
I built it successfully.
I confirmed `rosversion catkin_virtualenv` and showed 0.6.1.
However I got another problem and I'll make another thread.

closed 2020-09-05 by mqcmd196

mqcmd196, 2020-09-05:
I'm sorry, I just forgot to publish raw image from the camera. My problems have been completely solved. Thanks!