jsk-ros-pkg/coral_usb_ros #30: Error building on RP4 + Bionic + Melodic

agentsingh, 2020-09-11:
Hello,

Thanks for the great module. I was trying to build this on Raspberry Pi 4 with Ubuntu Bionic and Ros Melodic. I am getting the following error when executing catkin_make from ~/coral_ws. Can you please offer any guidance?

Run Build Command:"/usr/bin/make" "cmTC_81fe1/fast"
/usr/bin/make -f CMakeFiles/cmTC_81fe1.dir/build.make CMakeFiles/cmTC_81fe1.dir/build
make[1]: Entering directory '/home/agentsingh/coral_ws/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_81fe1.dir/CheckSymbolExists.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_81fe1.dir/CheckSymbolExists.c.o   -c /home/agentsingh/coral_ws/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_81fe1
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_81fe1.dir/link.txt --verbose=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_81fe1.dir/CheckSymbolExists.c.o  -o cmTC_81fe1
CMakeFiles/cmTC_81fe1.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x14): undefined reference to `pthread_create'
CheckSymbolExists.c:(.text+0x18): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_81fe1.dir/build.make:97: recipe for target 'cmTC_81fe1' failed
make[1]: *** [cmTC_81fe1] Error 1
make[1]: Leaving directory '/home/agentsingh/coral_ws/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_81fe1/fast' failed
make: *** [cmTC_81fe1/fast] Error 2

File /home/agentsingh/coral_ws/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/agentsingh/coral_ws/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_57447/fast"
/usr/bin/make -f CMakeFiles/cmTC_57447.dir/build.make CMakeFiles/cmTC_57447.dir/build
make[1]: Entering directory '/home/agentsingh/coral_ws/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_57447.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_57447.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.10/Modules/CheckFunctionExists.c
Linking C executable cmTC_57447
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_57447.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_57447.dir/CheckFunctionExists.c.o  -o cmTC_57447 -lpthreads
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_57447.dir/build.make:97: recipe for target 'cmTC_57447' failed
make[1]: *** [cmTC_57447] Error 1
make[1]: Leaving directory '/home/agentsingh/coral_ws/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_57447/fast' failed
make: *** [cmTC_57447/fast] Error 2

agentsingh, 2020-09-12:
I tried building again using catkin (instead of catkin_make) after installing catkin for melodic (sudo apt-get install ros-melodic-catkin). Given I am installing on Pi4 with Ubuntu Bionic, I made the following changes to the catkin config:

catkin config -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/aarch64-linux-gnu/libpython3.6m.so

I was able to build 18 out of 19 packages and then got the following error on running "catkin build". Please advice if you have any suggestions to fix this.

--- Error log:

Finished  <<< multi_map_server                                   [ 34.0 seconds ]                                                                                                                 
__________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << coral_usb:make /home/username/coral_ws/logs/coral_usb/build.make.001.log                                                                                                           
  ERROR: Command errored out with exit status 1:
   command: /home/username/coral_ws/build/coral_usb/venv/bin/python /home/username/coral_ws/build/coral_usb/venv/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmp5hicsu_8
       cwd: /tmp/pip-install-uxl5irzi/numpy
  Complete output (10 lines):
  Traceback (most recent call last):
    File "/home/username/coral_ws/build/coral_usb/venv/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
      main()
    File "/home/username/coral_ws/build/coral_usb/venv/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/home/username/coral_ws/build/coral_usb/venv/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py", line 108, in get_requires_for_build_wheel
      backend = _build_backend()
    File "/home/username/coral_ws/build/coral_usb/venv/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py", line 99, in _build_backend
      obj = getattr(obj, path_part)
  AttributeError: module 'setuptools.build_meta' has no attribute '__legacy__'
  ----------------------------------------
ERROR: Command errored out with exit status 1: /home/username/coral_ws/build/coral_usb/venv/bin/python /home/username/coral_ws/build/coral_usb/venv/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmp5hicsu_8 Check the logs for full command output.
Traceback (most recent call last):
  File "/home/username/coral_ws/src/catkin_virtualenv/catkin_virtualenv/scripts/venv_install", line 45, in <module>
    extra_pip_args=[arg for arg in extra_pip_args.split(" ") if arg != ""],
  File "/home/username/coral_ws/src/catkin_virtualenv/catkin_virtualenv/src/catkin_virtualenv/venv.py", line 88, in install
    run_command(command + ["-r", req], check=True)
  File "<string>", line 44, in run_command
  File "/usr/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/home/username/coral_ws/build/coral_usb/venv/bin/python', '-m', 'pip', 'install', '-qq', '--retries', '10', '--timeout', '30', '-r', '/home/username/coral_ws/src/coral_usb_ros/requirements.txt']' returned non-zero exit status 1.
make[2]: *** [venv/bin/activate] Error 1
make[1]: *** [CMakeFiles/coral_usb_generate_virtualenv.dir/all] Error 2
make: *** [all] Error 2
cd /home/username/coral_ws/build/coral_usb; catkin build --get-env coral_usb | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..................................................................................................................................................................................................
Failed     << coral_usb:make                                     [ Exited with code 2 ]                                                                                                           
Failed    <<< coral_usb                                          [ 1 minute and 31.1 seconds ]                                                                                                    
[build] Summary: 18 of 19 packages succeeded.                                                                                                                                                     
[build]   Ignored:   2 packages were skipped or are blacklisted.                                                                                                                                  
[build]   Warnings:  None.                                                                                                                                                                        
[build]   Abandoned: None.                                                                                                                                                                        
[build]   Failed:    1 packages failed.                                                                                                                                                           
[build] Runtime: 2 minutes and 0.6 seconds total.                                                                                                                                                 
[build] Note: Workspace packages have changed, please re-source setup files to use them.

agentsingh, 2020-09-12:
When I run:

/usr/bin/python3 -c "import setuptools.build_meta; print(setuptools.build_meta.__file__); print(setuptools.build_meta.__legacy__)"

I get the following output so my setuptools are correct:

<setuptools.build_meta._BuildMetaLegacyBackend object at 0xffffa45e46a0>

knorth55, 2020-09-15:
Sorry for replying you late.

I only checked with Ubuntu + AMD64 CPU.
I haven't try with ARM64 PU or Raspberry PI 4, so I have not encountered your error.
But there are several comments.
First, you should use `catkin-tools` for building the workspace.
please try `pip install catkin-tools -U` or something like that.
Next, clean up your workspace by removing all `devel`, `build`, `logs` and `.catkin_tools` dirs every time after your build.
Then, you should check your `python3` version with `python3 --version`.

agentsingh, 2020-09-17:
No problem! Thanks for taking a look. I was able to successfully build on Ubuntu + x86_64 as well but still getting error on Raspberry Pi 4. I used catkin in my latest build after installing catkin-tools. I am still getting an error (log below). I understand if the module is not supported on Raspberry Pi but wanted to check before trying other ways. Appreciate your help.
Python 3 version is 3.6.9 and changed catkin config per the first line below:
```
catkin config -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/aarch64-linux-gnu/libpython3.6m.so

Below is the tail of console with error:

``
Starting  >>> jsk_tools                                                                                                                                                                   
Starting  >>> coral_usb                                                                                                                                                                   
Finished  <<< jsk_data                                           [ 17.7 seconds ]                                                                                                         
Finished  <<< jsk_tools                                          [ 19.2 seconds ]                                                                                                         
Starting  >>> multi_map_server                                                                                                                                                            
Finished  <<< multi_map_server                                   [ 28.7 seconds ]                                                                                                         
__________________________________________________________________________________________________________________________________________________________________________________________
Errors     << coral_usb:make /home/agentsingh/coral_ws/logs/coral_usb/build.make.000.log                                                                                                   
  ERROR: Command errored out with exit status 1:
   command: /home/agentsingh/coral_ws/build/coral_usb/venv/bin/python /home/agentsingh/coral_ws/build/coral_usb/venv/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpyd6t8y97
       cwd: /tmp/pip-install-3d2rn6s_/numpy
  Complete output (10 lines):
  Traceback (most recent call last):
    File "/home/agentsingh/coral_ws/build/coral_usb/venv/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
      main()
    File "/home/agentsingh/coral_ws/build/coral_usb/venv/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/home/agentsingh/coral_ws/build/coral_usb/venv/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py", line 108, in get_requires_for_build_wheel
      backend = _build_backend()
    File "/home/agentsingh/coral_ws/build/coral_usb/venv/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py", line 99, in _build_backend
      obj = getattr(obj, path_part)
  AttributeError: module 'setuptools.build_meta' has no attribute '__legacy__'
  ----------------------------------------
ERROR: Command errored out with exit status 1: /home/agentsingh/coral_ws/build/coral_usb/venv/bin/python /home/agentsingh/coral_ws/build/coral_usb/venv/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpyd6t8y97 Check the logs for full command output.
Traceback (most recent call last):
  File "/home/agentsingh/coral_ws/src/catkin_virtualenv/catkin_virtualenv/scripts/venv_install", line 45, in <module>
    extra_pip_args=[arg for arg in extra_pip_args.split(" ") if arg != ""],
  File "/home/agentsingh/coral_ws/src/catkin_virtualenv/catkin_virtualenv/src/catkin_virtualenv/venv.py", line 88, in install
    run_command(command + ["-r", req], check=True)
  File "<string>", line 44, in run_command
  File "/usr/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/home/agentsingh/coral_ws/build/coral_usb/venv/bin/python', '-m', 'pip', 'install', '-qq', '--retries', '10', '--timeout', '30', '-r', '/home/agentsingh/coral_ws/src/coral_usb_ros/requirements.txt']' returned non-zero exit status 1.
make[2]: *** [venv/bin/activate] Error 1
make[2]: *** Deleting file 'venv/bin/activate'
make[1]: *** [CMakeFiles/coral_usb_generate_virtualenv.dir/all] Error 2
make: *** [all] Error 2
cd /home/agentsingh/coral_ws/build/coral_usb; catkin build --get-env coral_usb | catkin env -si  /usr/bin/make --jobserver-fds=3,4 -j; cd -
..........................................................................................................................................................................................
Failed     << coral_usb:make                                     [ Exited with code 2 ]                                                                                                   
Failed    <<< coral_usb                                          [ 2 minutes and 15.9 seconds ]                                                                                           
[... cut ...]
Traceback (most recent call last):
  File "/home/agentsingh/.local/lib/python3.6/site-packages/trollius/base_events.py", line 395, in __del__
  File "/home/agentsingh/.local/lib/python3.6/site-packages/trollius/unix_events.py", line 65, in close
  File "/home/agentsingh/.local/lib/python3.6/site-packages/trollius/unix_events.py", line 166, in remove_signal_handler
  File "/usr/lib/python3.6/signal.py", line 47, in signal
TypeError: signal handler must be signal.SIG_IGN, signal.SIG_DFL, or a callable object
``

knorth55, 2020-09-18:
@agentsingh 
according to this issue  https://github.com/pypa/setuptools/issues/1694, 
you are using `pip3>=19.0.2` and `setuptools<40.8.0`.
in this package, we use apt default pip and setuptools as `python3-pip` and `python3-setuptools`.
can you check the mismatch of these versions?
If there are mismatches, I recommend to use `apt` one, but if you want to use latest one, you can upgrade `setuptools` to latest, which may avoid the build error.

```
dpkg -l | grep python3-pip
pip3 -version
```

```
dpkg -l | grep python3-setuptools
pip3 freeze | grep setuptools
```

knorth55, 2020-10-05:
if you still have the problem, please re-open this issue.

closed 2020-10-05 by knorth55