jsk-ros-pkg/coral_usb_ros #31: オーギュメンテ―ション

k-okada, 2020-10-05:
@knorth55 トレーニングデータをサーバに送る際に自動でデータオーギュメンテ―ションできたら便利だな，と思ったんだけど，すでにそういう機能があったりするかな？

- 色の変換はAPCの時にいろいろやっていたからどっかにコードがありそう
- 画像の拡大縮小は問題なくできるんだろうか？
- 画像の回転は回転中心が重要だったりするのかな．

Cc: @miyabitane

knorth55, 2020-10-05:
@k-okada @MiyabiTane configsファイルにオーグメンテーションについては書かれています．
もともとのSSDの学習スクリプトのままですが，`random_holizontal_flip`をしています．
https://github.com/knorth55/coral_usb_ros/blob/master/training/labelme_voc/scripts/configs/pipeline_mobilenet_v2_ssd_retrain_last_few_layers.config#L135-L142

もとから何が用意されているかは，ここを読むと書いています．
https://github.com/tensorflow/models/blob/v1.12.0/research/object_detection/builders/preprocessor_builder.py

試してはいないですが，コードを読む限り，先に定義されているものとしては以下のようなものがあるようです．
```
    'normalize_image':
        preprocessor.normalize_image,
    'random_pixel_value_scale':
        preprocessor.random_pixel_value_scale,
    'random_image_scale':
        preprocessor.random_image_scale,
    'random_rgb_to_gray':
        preprocessor.random_rgb_to_gray,
    'random_adjust_brightness':
        preprocessor.random_adjust_brightness,
    'random_adjust_contrast':
        preprocessor.random_adjust_contrast,
    'random_adjust_hue':
        preprocessor.random_adjust_hue,
    'random_adjust_saturation':
        preprocessor.random_adjust_saturation,
    'random_distort_color':
        preprocessor.random_distort_color,
    'random_jitter_boxes':
        preprocessor.random_jitter_boxes,
    'random_crop_to_aspect_ratio':
        preprocessor.random_crop_to_aspect_ratio,
    'random_black_patches':
        preprocessor.random_black_patches,
    'rgb_to_gray':
        preprocessor.rgb_to_gray,
    'scale_boxes_to_pixel_coordinates': (
        preprocessor.scale_boxes_to_pixel_coordinates),
    'subtract_channel_mean':
        preprocessor.subtract_channel_mean,
    'convert_class_logits_to_softmax':
        preprocessor.convert_class_logits_to_softmax,
```
```
  if step_type == 'random_horizontal_flip':
  if step_type == 'random_vertical_flip':
  if step_type == 'random_rotation90':
  if step_type == 'random_crop_image':
  if step_type == 'random_pad_image':
  if step_type == 'random_crop_pad_image':
  if step_type == 'random_resize_method':
  if step_type == 'resize_image':
  if step_type == 'ssd_random_crop':
  if step_type == 'ssd_random_crop_pad':
  if step_type == 'ssd_random_crop_fixed_aspect_ratio':
  if step_type == 'ssd_random_crop_pad_fixed_aspect_ratio':
```

ちなみにですが，学習スクリプトとしては`tensorflow/models@v1.12.0`の以下にあるPythonスクリプトを走らせています．
https://github.com/tensorflow/models/blob/v1.12.0/research/object_detection/model_main.py

knorth55, 2020-10-05:
@k-okada @MiyabiTane
APCのときは`imgaug`をつかっていました．
https://github.com/aleju/imgaug

APCとかだと，ここらへんのコードでオーグメンテーションしていますが，tensorflowの流れにうまくつなげられるかはわかりません．
https://github.com/start-jsk/jsk_apc/blob/master/demos/grasp_fusion/grasp_fusion_lib/aug.py
https://github.com/start-jsk/jsk_apc/blob/master/demos/instance_occlsegm/instance_occlsegm_lib/aug.py

最近は`albumentations`などもあります．
https://github.com/albumentations-team/albumentations

k-okada, 2020-10-05:
なるほど，で，https://github.com/tensorflow/models/blob/v1.12.0/research/object_detection/builders/preprocessor_builder.pyにrandom_holizontal_flipが見つからないけど，どこを見たらある？random_rotateみたいなのがあれば，シンプルに良くなりそうだけど．

knorth55, 2020-10-05:
@k-okada `random_horizontal_flip`は以下の部分に書かれています．
https://github.com/tensorflow/models/blob/v1.12.0/research/object_detection/builders/preprocessor_builder.py#L142-L148
定義はここに書かれています．
https://github.com/tensorflow/models/blob/v1.12.0/research/object_detection/core/preprocessor.py#L443-L537

RGB値をいじって輝度や彩度いじる，回転する，スケールする，はシンプルに良くなりそうな気がします．

k-okada, 2020-10-06:
なるほど，で

https://github.com/knorth55/coral_usb_ros/blob/master/training/labelme_voc/run.sh

のオプションを切り替えできるようにするか，あるいは，たいがいの場合で有効な
オーギュメンテーションだったら，docker側をいじってその機能をデフォルトで使えるようにして，
「もう一回同じプログラム走らせたら，結果は良くなっているはずだから」という状況を作ってみましょう．

ちなみに，
https://github.com/tensorflow/models/blob/v1.12.0/research/object_detection/builders/preprocessor_builder.py#L158
をみると９０度回転はあるのかな．これでも良い気がします．

knorth55, 2020-10-13:
こんな感じでPRをつくってみました．
dlboxでも学習できることを確認しました．

https://github.com/knorth55/coral_usb_ros/pull/32

k-okada, 2020-10-13:
ナイス！ @MiyabiTane @Kanazawanaoaki たぶん今あるトレーニングスクリプト同じものを走らせたら勝手にオーギュメンテ―ションして学習してくれるはずなので，試してみましょう．

MiyabiTane, 2020-10-13:
試してみます、ありがとうございます！

MiyabiTane, 2020-10-17:
遅くなりましたが、学習を試してみたところエラーになってしまいました。
ログはこちらです。よろしくお願い致します。
https://gist.github.com/MiyabiTane/70f1ed4569aa276dfaf186bfcd6dd940#file-coral-L3090-L3133
（num_training_stepsはデフォルトで2000だったのに指定してしまいました...）

knorth55, 2020-10-17:
@MiyabiTane 
これはおそらく @tongtybj 先生がdlbox1で学習をしていたので，GPUメモリが足りずに学習できなかったようですね．
学習するまえにnvidia-smiでGPUのメモリが使われているかどうかをチェックしてみるといいかとおもいます．
同じようにdlbox2やdlbox3にログインできると思うので，そちらでやってみてください．
```
$ nvidia-smi
Sat Oct 17 23:39:33 2020
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 410.129      Driver Version: 410.129      CUDA Version: 10.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  TITAN X (Pascal)    Off  | 00000000:05:00.0  On |                  N/A |
| 39%   66C    P2    97W / 250W |  12180MiB / 12192MiB |     67%      Default |
+-------------------------------+----------------------+----------------------+
|   1  TITAN X (Pascal)    Off  | 00000000:06:00.0 Off |                  N/A |
| 49%   81C    P2   105W / 250W |  11679MiB / 12196MiB |     66%      Default |
+-------------------------------+----------------------+----------------------+
|   2  GeForce GTX 108...  Off  | 00000000:09:00.0 Off |                  N/A |
| 76%   82C    P2    98W / 250W |   9555MiB / 11178MiB |     65%      Default |
+-------------------------------+----------------------+----------------------+
|   3  GeForce GTX 108...  Off  | 00000000:0A:00.0 Off |                  N/A |
| 65%   67C    P2   101W / 280W |  10197MiB / 11178MiB |     75%      Default |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      7273      G   /usr/lib/xorg/Xorg                            39MiB |
|    0     16539      C   /opt/conda/bin/python                      12129MiB |
|    1     16540      C   /opt/conda/bin/python                      11667MiB |
|    2     16541      C   /opt/conda/bin/python                       9543MiB |
|    3     16542      C   /opt/conda/bin/python                      10185MiB |
+-----------------------------------------------------------------------------+
```

また金沢くんが別のGPUを選んで学習できるようにスクリプトをかえてくれました． #35
これを使うと同じDlboxの違うGPUでも学習できると思います．

cc. @Kanazawanaoaki

tongtybj, 2020-10-17:
@knorth55 @MiyabiTane 

すみません。告知なしでdlbox1をフルで使っています。
もうしばらく(一ヶ月ぐらい?)、使う予定ですが、
これはメールで告知した方がいいのかな？ @knorth55

MiyabiTane, 2020-10-17:
dlbox2で試してみたのですが、同じエラーになってしまいました。
```
tanemoto@dlbox2:~$ nvidia-smi
Sun Oct 18 00:27:42 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 410.72       Driver Version: 410.72       CUDA Version: 10.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  TITAN X (Pascal)    Off  | 00000000:05:00.0 Off |                  N/A |
| 18%   31C    P0    59W / 250W |      0MiB / 12194MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   1  TITAN X (Pascal)    Off  | 00000000:06:00.0 Off |                  N/A |
| 18%   34C    P0    63W / 250W |      0MiB / 12196MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   2  TITAN X (Pascal)    Off  | 00000000:09:00.0 Off |                  N/A |
| 34%   50C    P0    58W / 250W |      0MiB / 12196MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   3  TITAN X (Pascal)    Off  | 00000000:0A:00.0 Off |                  N/A |
| 26%   41C    P0    57W / 250W |      0MiB / 12196MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+

```

MiyabiTane, 2020-10-17:
貼り忘れましたすみません
https://gist.github.com/MiyabiTane/bad1bcc18cf45db2a7dc3a54e2f9d6eb

knorth55, 2020-10-17:
>すみません。告知なしでdlbox1をフルで使っています。
もうしばらく(一ヶ月ぐらい?)、使う予定ですが、
これはメールで告知した方がいいのかな？ @knorth55

@tongtybj 
そうですね，告知してもらうと嬉しいです．
もしくはdlbox1, dlbox2は避けてもらえたらと思います．
B4の人たちはdlbox1, 2の桁が少ないところをよく使っているイメージなので．

tongtybj, 2020-10-17:
@knorth55 

了解です。
dlbox7,8あたりを使い、告知するようにします。

knorth55, 2020-10-17:
@tongtybj よろしくお願いします．

knorth55, 2020-10-17:
@MiyabiTane dlbox2で今さっき学習してみましたができました．つかっているrun.shは`coral_usb_ros/training/label_me_voc/run.sh`ですか？データセットも共有してくれると嬉しいです．

tongtybj, 2020-10-17:
@knorth55 @MiyabiTane 

dlbox1の方を解放しました。

MiyabiTane, 2020-10-18:
> @MiyabiTane dlbox2で今さっき学習してみましたができました．つかっているrun.shはcoral_usb_ros/training/label_me_voc/run.shですか？データセットも共有してくれると嬉しいです

@tongtybjありがとうございます。 
@knorth55 試して頂きありがとうございます。coral_usb_ros/training/label_me_voc/run.shです。データセットは以下のtrainとtestです。https://drive.google.com/drive/folders/1pVdDRnUVv1WZ7sSrFfaX1DYGBjwvbiOy

MiyabiTane, 2020-10-18:
今朝dlbox1でやり直したところ学習させることができました。お騒がせしました。

MiyabiTane, 2020-10-18:
オーギュメンテーション前と後での認識の様子の動画を撮ったので共有させて頂きます。
オーギュメンテーション前▷https://drive.google.com/file/d/1xjAtXM2X58vmX8dtt4wF4tTApu9pfvEo/view?usp=sharing
オーギュメンテーション後▷https://drive.google.com/file/d/1WvVLZbHghNnx6tPO9W7sV9qd5bEB_WYk/view?usp=sharing
学習結果のグラフや入力画像はここにまとめてあります。
https://drive.google.com/drive/folders/18zGhtHVP6BRTxX_mL1gOnJ9Oj9Iaw073
20201001▷num_training_steps2000と20201018を比較して頂ければと思います。

knorth55, 2020-10-18:
これはかなりうまくいってそうな感じですかね！
Trackingもパパッとできればさらに安定しそうですね
GPUいらないTrackingは準備してみます

MiyabiTane, 2020-10-18:
だいぶ良くなった印象です！
ありがとうございます！