jsk-ros-pkg/coral_usb_ros #53: dynamic_reconfigure goes wrong when edgetpu related nodes exist

ishiguroJSK, 2021-01-27:
When the target ROS network has edgetpu related nodes and I set my rosmaster connected to the machine, I got these error.

I checked with pr1040 or my docker environment.
It seems to occur both on 16.04 and 18.04.

```
($ rossetmaster pr1040 && rossetip)
$ rosrun rqt_reconfigure rqt_reconfigure 
PluginManager._load_plugin() could not load plugin "rqt_reconfigure/Param":
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 102, in load
    self._load()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 55, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 101, in load
    return class_ref(plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/param_plugin.py", line 51, in __init__
    self._plugin_widget = ParamWidget(context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/param_widget.py", line 105, in __init__
    self, rp, self.sig_sysmsg
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/node_selector_widget.py", line 102, in __init__
    self._update_nodetree_pernode()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/node_selector_widget.py", line 341, in _update_nodetree_pernode
    TreenodeQstdItem.NODE_FULLPATH
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/treenode_qstditem.py", line 94, in __init__
    self._set_param_name(grn_current_treenode)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/treenode_qstditem.py", line 246, in _set_param_name
    self._toplevel_treenode_name = self._list_treenode_names[0]
IndexError: list index out of range
```

knorth55, 2021-01-27:
when you run the `rqt_reconfigure`, did you source `coral` workspace?
can you try the same thing with sourcing coral workspace?

ishiguroJSK, 2021-01-27:
Whether I did `source ~/coral_ws/devel/setup.bash ` or `source /opt/ros/kinetic/setup.bash`, It happens.

ishiguroJSK, 2021-01-27:
It also happens on the other PCs in 73B2

knorth55, 2021-02-05:
@ishiguroJSK I fix this issue in https://github.com/knorth55/coral_usb_ros/commit/8a480a01ca1708f0e5445e1dc5442d74d8b18e7a and applied to PR1040.
Can you check if rqt_reconfigure is working fine again?

ishiguroJSK, 2021-02-05:
LGTM！！
Thank you

closed 2021-02-05 by ishiguroJSK