jsk-ros-pkg/coral_usb_ros #66: no jsk_perception file

a-ichikura, 2021-08-13:
I tried to do this according to README.
It seemed that installing the Edge TPU runtime and the TensorFlow Lite interpreter(Melodic) succeeded(There weren't any errors).

However, I can't run insta360 air for panorama image though image_publisher and usb_cam is successfully running. 

The error message is
`RLException: [sample_insta360_air.launch] is neither a launch file in package [jsk_perception] nor is [jsk_perception] a launch file name
The traceback for the exception was written to the log file`
 
In addition, Edge TPU launch files do not work.

`source /opt/ros/melodic/setup.bash
source ~/coral_ws/devel/setup.bash
roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=/image_publisher/output`

Even though I tried to run image view in normal workspace(not edge tpu workspace),it did not work.

How can I solve this issue?

knorth55, 2021-08-13:
can you paste all the output of the error log when you run `roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=/image_publisher/output`?

knorth55, 2021-08-13:
if you want to run insta360, you need insta360 panorama camera, but do you have it?

a-ichikura, 2021-08-13:
This is all message when I run the code.
`RLException: [edgetpu_object_detector.launch] is neither a launch file in package [coral_usb] nor is [coral_usb] a launch file name
The traceback for the exception was written to the log file
`

knorth55, 2021-08-13:
it seems you didn't do `source ~/coral_ws/devel/setup.bash`.
source your workspace and try again.

and please copy and paset whole output lines. all. everything.

knorth55, 2021-08-13:
can you try these codes one buy one carefully? and put all output?

```bash
source /opt/ros/melodic/setup.bash
rosrun usb_cam usb_cam_node
```

```bash
source ~/coral_ws/devel/setup.bash
roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=/usb_cam/image_raw
```

```bash
source /opt/ros/kinetic/setup.bash
rosrun image_view image_view image:=/edgetpu_object_detector/output/image
```

a-ichikura, 2021-08-13:
leus@ubuntu:~/coral_ws/src/coral_usb_ros/scripts$ roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=/image_publisher/output
... logging to /home/leus/.ros/log/24476b72-fc2c-11eb-83fa-18cc1878776f/roslaunch-ubuntu-5635.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.10.106:39681/

SUMMARY
========

PARAMETERS
 * /edgetpu_object_detector/device_id: 0
 * /edgetpu_object_detector/image_transport: raw
 * /edgetpu_object_detector/label_file: /home/leus/coral_...
 * /edgetpu_object_detector/model_file: /home/leus/coral_...
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    edgetpu_object_detector (coral_usb/edgetpu_object_detector.py)

ROS_MASTER_URI=http://localhost:11311

process[edgetpu_object_detector-1]: started with pid [5650]
[INFO] [1628859599.086150]: Using transport raw
[WARN] [1628859604.095315]: [/edgetpu_object_detector] subscribes topics only with child subscribers. Set '~always_subscribe' as True to have it subscribe always.

knorth55, 2021-08-13:
with the output, it seems working correctly.
does your pc have camera?

a-ichikura, 2021-08-13:
yes, I have a camera.

knorth55, 2021-08-13:
so how does image_view look like?
does it shows image?

a-ichikura, 2021-08-13:
leus@ubuntu:~/coral_ws/src/coral_usb_ros/scripts$ source /opt/ros/melodic/setup.bash
leus@ubuntu:~/coral_ws/src/coral_usb_ros/scripts$ rosrun image_view image_view image:=/edgetpu_object_detector/output/image
[ INFO] [1628860414.476830763]: Initializing nodelet with 8 worker threads.
[ INFO] [1628860414.563775504]: Using transport "raw"

knorth55, 2021-08-13:
I update the readme.
you can check the output image with `image_view` .
https://github.com/knorth55/coral_usb_ros#run-image_view

knorth55, 2021-08-13:
how does GUI looks like?

knorth55, 2021-08-13:
can you also check the frequency of the output?
```
rostopic hz /edgetpu_object_detector/output/image
```

a-ichikura, 2021-08-13:
no window appears

knorth55, 2021-08-13:
thats wierd.
how about `rosrun rqt_image_view rqt_image_view`?

a-ichikura, 2021-08-13:
I checked the frequency

leus@ubuntu:~/coral_ws/src/coral_usb_ros/scripts$ rostopic hz /edgetpu_object/output/image
WARNING: topic [/edgetpu_object/output/image] does not appear to be published yet

knorth55, 2021-08-13:
can you paste the output of this command?
```
rosnode list
```
```
rostopic list
```
```
rosnode info /edgetpu_object_detector
```

```
rosnode ping /edgetpu_object_detector
```

```
rostopic info /edgetpu_object_detector/output/image
```

```
rostopic hz /edgetpu_object_detector/output/image
```

knorth55, 2021-08-13:
>rostopic hz /edgetpu_object/output/image

also you are mistyping.

a-ichikura, 2021-08-13:
leus@ubuntu:~/coral_ws/src/coral_usb_ros/scripts$ rosnode list
/image_view
/rosout
/rqt_gui_cpp_node_6856

leus@ubuntu:~/coral_ws/src/coral_usb_ros/scripts$ rostopic list
/edgetpu_object_detector/output/image
/image_view/output
/image_view/parameter_descriptions
/image_view/parameter_updates
/rosout
/rosout_agg

a-ichikura, 2021-08-13:
leus@ubuntu:~/coral_ws/src/coral_usb_ros/scripts$ rosnode info /edgetpu_object_detector
Node [/edgetpu_object_detector]
Publications: None

Subscriptions: None

Services: None

leus@ubuntu:~/coral_ws/src/coral_usb_ros/scripts$ rosnode ping /edgetpu_object_detector
rosnode: node is [/edgetpu_object_detector]
cannot ping [/edgetpu_object_detector]: unknown node

a-ichikura, 2021-08-13:
leus@ubuntu:~/coral_ws/src/coral_usb_ros/scripts$ rostopic info /edgetpu_object_detector/output/image
Type: sensor_msgs/Image

Publishers: None

Subscribers: 
 * /image_view (http://192.168.10.106:38761/)

leus@ubuntu:~/coral_ws/src/coral_usb_ros/scripts$ rostopic hz /edgetpu_object_detector/output/image
subscribed to [/edgetpu_object_detector/output/image]
no new messages
no new messages
no new messages

knorth55, 2021-08-13:
>leus@ubuntu:~/coral_ws/src/coral_usb_ros/scripts$ rosnode list
/image_view
/rosout
/rqt_gui_cpp_node_6856

are you keep running this launch?

```
source ~/coral_ws/devel/setup.bash
roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=/usb_cam/image_raw
```

a-ichikura, 2021-08-13:
sorry,  I am running image_view. I relaunch.

leus@ubuntu:~/coral_ws/src/coral_usb_ros/scripts$ rostopic list
/edgetpu_object_detector/output/class
/edgetpu_object_detector/output/image
/edgetpu_object_detector/output/image/compressed
/edgetpu_object_detector/output/rects
/edgetpu_object_detector/parameter_descriptions
/edgetpu_object_detector/parameter_updates
/rosout
/rosout_agg
leus@ubuntu:~/coral_ws/src/coral_usb_ros/scripts$ rosnode ping /edgetpu_object_detector
rosnode: node is [/edgetpu_object_detector]
pinging /edgetpu_object_detector with a timeout of 3.0s
xmlrpc reply from http://192.168.10.106:36579/	time=0.826120ms
xmlrpc reply from http://192.168.10.106:36579/	time=2.608061ms
xmlrpc reply from http://192.168.10.106:36579/	time=2.347946ms

knorth55, 2021-08-13:
please run these three commands in different terminal.
https://github.com/knorth55/coral_usb_ros/issues/66#issuecomment-898442658

a-ichikura, 2021-08-13:
> please run these three commands in different terminals in parallel
> [#66 (comment)](https://github.com/knorth55/coral_usb_ros/issues/66#issuecomment-898442658)

Succeeded.

knorth55, 2021-08-13:
Good!

knorth55, 2021-08-13:
if you want to try with another image topic (like robot image topic), you can do like this.

```bash
source /opt/ros/melodic/setup.bash
rossetip
rossetmaster <robot ip>
source ~/coral_ws/devel/setup.bash
roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=<robot image topic name>
```

```bash
source /opt/ros/melodic/setup.bash
rossetip
rossetmaster <robot ip>
rosrun image_view image_view image:=/edgetpu_object_detector/output/image
```