jsk-ros-pkg/coral_usb_ros #72: Tuple index out of range

sugihara-16, 2021-11-21:
I tried to run the launch file named ''edgetpu_object_detector.launch" by following processes, 

```
# terminal 1
$ roscore

# terminal 2
$ source /opt/ros/melodic/setup.bash
$ rosrun usb_cam usb_cam_node

# terminal 3
$ source /opt/ros/${ROS_DISTRO}/setup.bash
$ source ~/coral_ws/devel/setup.bash
$ roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=/usb_cam/image_raw
```

and I got this error

```
SUMMARY
========

PARAMETERS
 * /edgetpu_object_detector/device_id: 0
 * /edgetpu_object_detector/image_transport: raw
 * /edgetpu_object_detector/label_file: /home/mech-user/c...
 * /edgetpu_object_detector/model_file: /home/mech-user/c...
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    edgetpu_object_detector (coral_usb/edgetpu_object_detector.py)

ROS_MASTER_URI=http://localhost:11311

process[edgetpu_object_detector-1]: started with pid [21442]
[INFO] [1637505613.824623]: Using transport raw
Traceback (most recent call last):
  File "/home/mech-user/coral_ws/src/coral_usb_ros/node_scripts/edgetpu_object_detector.py", line 10, in <module>
    detector = EdgeTPUObjectDetector()
  File "/home/mech-user/coral_ws/src/coral_usb_ros/python/coral_usb/object_detector.py", line 11, in __init__
    super(EdgeTPUObjectDetector, self).__init__(None, None, namespace)
  File "/home/mech-user/coral_ws/src/coral_usb_ros/python/coral_usb/detector_base.py", line 68, in __init__
    device_path = ListEdgeTpuPaths(EDGE_TPU_STATE_NONE)[device_id]
IndexError: tuple index out of range
[edgetpu_object_detector-1] process has died [pid 21442, exit code 1, cmd /home/mech-user/coral_ws/devel/lib/coral_usb/edgetpu_object_detector.py ~input:=/usb_cam/image_raw __name:=edgetpu_object_detector __log:=/home/mech-user/.ros/log/ece972ba-4ad7-11ec-b47e-2079185f7c3d/edgetpu_object_detector-1.log].
log file: /home/mech-user/.ros/log/ece972ba-4ad7-11ec-b47e-2079185f7c3d/edgetpu_object_detector-1*.log

```

Could anyone can help me?

knorth55, 2021-11-21:
it seems your pc does not recognize TPU USB.
Can you check if you installed coral TPU USB correctly?
or can you unplug and plug the TPU USB or reboot your pc?

sugihara-16, 2021-11-22:
I appreciate your help!
After reboot my pc, it went well!

closed 2021-12-23 by knorth55