jsk-ros-pkg/coral_usb_ros #81: noetic-jsk-perception failed

slabban, 2022-02-09:
Hello!

I'm following the noetic installation instructions and am running into this error while building a docker image :
    #17 1117.7 executing command [sudo -H apt-get install -y ros-noetic-jsk-perception]
    #17 1117.7 executing command [sudo -H apt-get install -y ros-noetic-resource-retriever]
    #17 1117.7 executing command [sudo -H apt-get install -y ros-noetic-catkin-virtualenv]
    #17 1117.7 executing command [sudo -H apt-get install -y ros-noetic-cv-bridge]
    #17 1117.7 executing command [sudo -H apt-get install -y ros-noetic-jsk-recognition-msgs]
    #17 1117.7 executing command [sudo -H apt-get install -y ros-noetic-jsk-topic-tools]
    #17 1117.7 ERROR: the following rosdeps failed to install
    #17 1117.7   apt: command [sudo -H apt-get install -y ros-noetic-jsk-perception] failed
    #17 1117.7   apt: Failed to detect successful installation of [ros-noetic-jsk-perception]

I have added a 'rosdep update' to your instructions since it was needed on my end since it was needed for my system to work:
    cd ~/coral_ws/src/ && \
    rosdep update -y && \
    rosdep install --from-paths . --ignore-src -y -r && \

knorth55, 2022-02-10:
Hmm, this is weird.
`jsk_perception` is available in noetic `ros` and `ros-testing`.
http://packages.ros.org/ros/ubuntu/pool/main/r/ros-noetic-jsk-perception/
http://packages.ros.org/ros-testing/ubuntu/pool/main/r/ros-noetic-jsk-perception/
https://build.ros.org/job/Nbin_uF64__jsk_perception__ubuntu_focal_amd64__binary/

can you do `sudo apt update` before you run the command?

slabban, 2022-02-10:
Looks like that did the trick.. a valuable lesson on Docker best practices to include 'apt-get update' in every RUN command that has 'install'.. apparently its called 'cache busting':
https://docs.docker.com/develop/develop-images/dockerfile_best-practices/

Thank you for the solid suggestion.. I saw you name in the Jsk repo and figured you would have a solution.

knorth55, 2022-02-10:
Good to hear that!

knorth55, 2022-02-10:
I also recommend to remove the cache to reduce your image size.
So I think it is better to adding `apt-get update` at the top and `rm /var/lib/apt/lists/*` at the bottom for every `RUN` command.
https://docs.docker.com/develop/develop-images/dockerfile_best-practices/#run

slabban, 2022-02-10:
That's an interesting catch.  Though It says at the bottom of that section 'Official Debian and Ubuntu images automatically run apt-get clean so explicit invocation is not required. ', and the ros noetic image I am using is ultimately built on Ubuntu Focal:
https://github.com/osrf/docker_images/blob/11c613986e35a1f36fd0fa18b49173e0c564cf1d/ros/noetic/ubuntu/focal/ros-core/Dockerfile

I wonder why it is still being run.. best practice?

knorth55, 2022-02-10:
Official images remove the apt cache in there `Dockerfile`, but it does not have a effect on latter image.
Even if official image remove the apt, latrer image or layer uses apt, it will keep apt cache.
So if your image on `ubuntu`, install something and do not remove the apt cache, it will keep the cache.
It will use old apt cache because docker skip already build layer to speed up build process by default.
you can force build again with flag something like `--no-cache`.

slabban, 2022-02-11:
Thank you for the above advice, already implementing it!

I reopened the issue due the jsk-perception packages not being available in the ubuntu pool for armhf, I am building this image on a Raspberry Pi 4 Buster OS. Tried cloning the jsk-recognition into the workspace and building it locally but still no success.

I can install an Ubuntu Focal arm64 Server Image since it's proven to work there:
https://cdimage.ubuntu.com/releases/20.04/release/

Before I take that step, wanted to know if you have any workaround for the current armhf configuration.

knorth55, 2022-02-11:
Armhf + Debian Buster is not supported for ROS debian packages.
It is quite complicated, but
Armhf + Ubuntu Focal is supported but Armhf + Debian Buster is not.
It is written in REP3.
https://www.ros.org/reps/rep-0003.html#noetic-ninjemys-may-2020-may-2025

So if you want to do it, you need to build it from source.
But i recommend to install Ubuntu Mate, and install it from deb package.

slabban, 2022-02-15:
That really insightful stuff, I appreciate that again.

To confirm, everything is working fine now. Ubuntu installation was the way to go!

closed 2022-02-15 by slabban