jsk-ros-pkg/coral_usb_ros #90: rviz support ?

Petros626, 2022-03-15:
hey,

does anyone know if it is possible to use this ros package **coral_usb_ros** together with rviz to display data from a LiDAR sensor on which object detection is performed?

thanks for any help and advice

knorth55, 2022-03-15:
yes, you can use object detection data with point cloud data.
we connect this package to `jsk_recognition` and get the bounding box as follow.
https://github.com/jsk-ros-pkg/jsk_recognition/pull/2513

Petros626, 2022-03-15:
hey thanks for your answer.

I am currently doing my bachelor thesis and will probably need your repo later, so I thought I would inform myself.

I first have to read up on the repo and see exactly how it works. So, to avoid any misunderstandings, with the ROS I can
 
Do I understand correctly that the coral_usb_ros package interprets the custom TFLite model and with jsk_recognition the bounding boxes are created in rviz? Is it possible to display them as 2D bounding boxes? Maybe by changing the ROS message: vision_msgs/BoundingBox2D.msg

knorth55, 2022-03-16:
You can train with your own dataset and use custom model.
Please read the following section.
https://github.com/knorth55/coral_usb_ros/tree/master/training

`vision_msgs` is a new repo, so we don't support it and use `jsk_recognition_msgs`.
But you can visualize the bounding box on rviz, if you write the message conversion script.
I don't know how to visualize 2d bounding box on rviz 3d plane, so please find how to do it.

Petros626, 2022-03-16:
thanks I have already looked at that, I will do the training without ROS but with Anaconda. When I get to the point of integrating object recognition into ROS, respectively my TFLite model. I'll let you know as soon as I know more.

Thanks a lot so far

Petros626, 2022-05-12:
@knorth55  Hey,

In the meantime, I have moved on and would now like to address the fact that this node can be used with point cloud data. the node only accepts sensor_msgs/Image, but my point cloud data has the format sensor_msgs/PointCloud2, so how was that meant then? Was this about cameras that generate point clouds, such as kinect?

knorth55, 2022-05-12:
you cannot use pointcloud as input.
you need to change from pointcloud to image and use the image as input.

Petros626, 2022-05-12:
> you cannot use pointcloud as input. you need to change from pointcloud to image and use the image as input.

yes this was exactly what i thought. Any sources/nodes to do so?

knorth55, 2022-05-12:
https://wiki.ros.org/pcl_ros/Tutorials/CloudToImage

Petros626, 2022-05-13:
> https://wiki.ros.org/pcl_ros/Tutorials/CloudToImage

yeah i've found this too, but its not working for unexplained reasons. I mean just copy the code and run it with node, which published cloud message

Petros626, 2022-05-14:
Yes I've checked this, it's only for Cameras like Kinect. Not for LiDAR sensors :)

knorth55, 2022-05-14:
in order to run `CloudToImage`, point cloud should be organized.
`is_dense` should be `false`.
https://www.mathworks.com/help/lidar/ug/organized-and-unorganized-point-clouds.html#:~:text=An%20organized%20point%20cloud%20resembles,xyz%2Dcoordinates%20of%20its%20points.

knorth55, 2022-05-14:
you should use RGB image instead of LiDAR sensor.

Petros626, 2022-05-14:
> in order to run `CloudToImage`, point cloud should be organized. `is_dense` should be `false`. https://www.mathworks.com/help/lidar/ug/organized-and-unorganized-point-clouds.html#:~:text=An%20organized%20point%20cloud%20resembles,xyz%2Dcoordinates%20of%20its%20points.

the parameter `is_dense` is` True` in my case. Yes i convert LiDAR PointCloud with a node to a OpenCV BEV Image. The channels are switched, so probably should conert it from BGR to RGB before using your node. Could be possible it doesn't matter, because I get pictures only with white (objects) and black pixels (environment,room)

closed 2022-07-12 by knorth55