jsk-ros-pkg/coral_usb_ros #98: catkin build hangs

joeuser846, 2022-05-15:
I'm trying to install this code under Ubuntu 20.04.4 LTS and ROS Noetic on a Raspberry Pi 4.  When I get to the catkin build step the  process always hangs at 84%.  Any ideas?  Thanks.

knorth55, 2022-05-15:
does you raspberrypi have internet connection?
it need internet to install dependencies.
or if it is memory usage problem, you can change the thread number with `catkin build -j 2`

joeuser846, 2022-05-15:
Thanks for your good suggestion, but that didn't turn-out to be the problem.  I decided to do 'catkin clean', 'catkin init', and 'catkin build' and that resulted in hanging at 84% for about 60 secs, but then it finished with two errors:  'testresources' missing and 'numpy' wrong version installed.  I resolved those errors with:

sudo apt-get install -y python3-testresources

pip3 uninstall numpy
pip3 install numpy==1.19.2

The 'catkin build' now completes.

closed 2022-05-16 by knorth55