jsk-ros-pkg/coral_usb_ros #99: ImportError: No module named '_edgetpu_cpp_wrapper'

joeuser846, 2022-05-16:
I'm trying to install this node on the ElementaryOS derivative of Ubuntu 20:

Linux ElementaryOS 5.13.0-41-generic #46~20.04.1-Ubuntu SMP Wed Apr 20 13:16:21 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
with
ROS Noetic and Python 3.8.10 (system version)

When I try to launch the object detection node with:

roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=/usb_cam/image_raw

I get the following error about a missing python module.  Any suggestions are appreciated.

Traceback (most recent call last):
  File "/home/user/Development/coral_ws/src/coral_usb_ros/node_scripts/edgetpu_object_detector.py", line 5, in <module>
    from coral_usb.object_detector import EdgeTPUObjectDetector
  File "/home/user/Development/coral_ws/src/coral_usb_ros/python/coral_usb/object_detector.py", line 6, in <module>
    from coral_usb.detector_base import EdgeTPUDetectorBase
  File "/home/user/Development/coral_ws/src/coral_usb_ros/python/coral_usb/detector_base.py", line 45, in <module>
    from edgetpu.basic.edgetpu_utils import EDGE_TPU_STATE_ASSIGNED
  File "/usr/local/lib/python3.8/dist-packages/edgetpu/basic/edgetpu_utils.py", line 15, in <module>
    import edgetpu.swig.edgetpu_cpp_wrapper as wrapper
  File "/usr/local/lib/python3.8/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 28, in <module>
    _edgetpu_cpp_wrapper = swig_import_helper()
  File "/usr/local/lib/python3.8/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 20, in swig_import_helper
    import _edgetpu_cpp_wrapper
ModuleNotFoundError: No module named '_edgetpu_cpp_wrapper'

knorth55, 2022-05-16:
did you install `python3-edgetpu`?
it seems you install `edgetpu` with `sudo pip` or something because it is installed in `/usr/local/lib/python3.8`.
please follow the step written in readme.
https://github.com/jsk-ros-pkg/coral_usb_ros#edge-tpu-dependencies-installation

joeuser846, 2022-05-16:
Thanks for your help again.  I discovered that the library was installed in the correct place but wasn't being found.  I fixed it by extending PYTHONPATH to include that directory.  Then I got two more missing library errors that I fixed with:

sudo apt install ros-noetic-jsk-topic-tools
sudo apt install ros-noetic-jsk-recognition-msgs

I'm not sure how my system got into that state, but it's working now.  Perhaps it happened because I followed the instructions on the Coral site for setup before finding your repo?

closed 2022-06-02 by knorth55