jsk-ros-pkg/coral_usb_ros #100: tensorflow model support

Petros626, 2022-05-25:
Hey,

I've read that only ssd models are supported for TFLite1 and TFLite2, so the hardware accelerator (Edge TPU) supports only ssd models until now. I saw that in the example the ssd_mobilenet_v2 i used for inference. But would the newer one (SSD MobileNet V2 FPNLite 640x640) of TensorFlow 2 also with this node?

knorth55, 2022-05-25:
Only tensorflow1 is supported

Petros626, 2022-05-25:
Okay thank you. So you dont know it 100%. Is there the option to ask the guy, who added the support for ssd_mobilenet_v2?

knorth55, 2022-05-25:
Sorry, but I dont understand your question.
Mobilenet V2 is the model name, and it doesnt mean the model is for tensorflow 2

Petros626, 2022-05-28:
Yeah thank you for the detailed explanation, if I have the time I would do so, but I'm working on my bachelor degree, so for first use it should only run on TensorFlow 1 model. 

Hahah I think the guys from TensorFlow do their best :D