jsk-ros-pkg/coral_usb_ros #110: /edgetpu_object_detector/output/image is not published

708yamaguchi, 2022-06-07:
When I use fetch15, `/edgetpu_object_detector/output/image` is not published.

```
fetch@fetch15:~/ros/melodic/src/jsk-ros-pkg/coral_usb_ros$ rostopic hz /edgetpu_object_detector/output/image
subscribed to [/edgetpu_object_detector/output/image]
no new messages
no new messages
```

EdgeTPU is recognized
```
fetch@fetch15:~/ros/melodic/src/jsk-ros-pkg/coral_usb_ros$ lsusb
Bus 005 Device 008: ID 18d1:9302 Google Inc. 
Bus 005 Device 004: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 005 Device 007: ID 8087:0b37 Intel Corp. 
Bus 005 Device 003: ID 18d1:9302 Google Inc. 
Bus 005 Device 002: ID 2109:0817 VIA Labs, Inc. 
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 013: ID 2886:0018  
Bus 004 Device 011: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 004 Device 009: ID 1130:1620 Tenx Technology, Inc. 
Bus 004 Device 006: ID 2109:2817 VIA Labs, Inc. 
Bus 004 Device 005: ID 2e1a:1000  
Bus 004 Device 004: ID 222a:004d  
Bus 004 Device 012: ID 8087:0aa7 Intel Corp. 
Bus 004 Device 010: ID 26ce:01a2  
Bus 004 Device 008: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 004 Device 002: ID 0d8c:0126 C-Media Electronics, Inc. 
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 1d27:0609 ASUS 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

The node manager is alive and there seems to be no error.
```
fetch@fetch15:~/ros/melodic/src/jsk-ros-pkg/coral_usb_ros$ rostopic info /edgetpu_object_detector/output/image
Type: sensor_msgs/Image

Publishers: 
 * /edgetpu_object_node_manager (http://fetch15:37505/)

Subscribers: None
```

```
fetch@fetch15:~/ros/melodic/src/jsk-ros-pkg/coral_usb_ros$ rosnode ping /edgetpu_object_node_manager
rosnode: node is [/edgetpu_object_node_manager]
pinging /edgetpu_object_node_manager with a timeout of 3.0s
xmlrpc reply from http://fetch15:37505/	time=0.869989ms
xmlrpc reply from http://fetch15:37505/	time=0.595093ms
```

```
fetch@fetch15:~/ros/melodic/src/jsk-ros-pkg/coral_usb_ros$ rosnode info /edgetpu_object_node_manager
--------------------------------------------------------------------------------
Node [/edgetpu_object_node_manager]
Publications: 
 * /diagnostics [diagnostic_msgs/DiagnosticArray]
 * /edgetpu_object_detector/output/class [jsk_recognition_msgs/ClassificationResult]
 * /edgetpu_object_detector/output/image [sensor_msgs/Image]
 * /edgetpu_object_detector/output/image/compressed [sensor_msgs/CompressedImage]
 * /edgetpu_object_detector/output/rects [jsk_recognition_msgs/RectArray]
 * /edgetpu_object_detector/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /edgetpu_object_detector/parameter_updates [dynamic_reconfigure/Config]
 * /edgetpu_panorama_object_detector/output/class [jsk_recognition_msgs/ClassificationResult]
 * /edgetpu_panorama_object_detector/output/image [sensor_msgs/Image]
 * /edgetpu_panorama_object_detector/output/image/compressed [sensor_msgs/CompressedImage]
 * /edgetpu_panorama_object_detector/output/rects [jsk_recognition_msgs/RectArray]
 * /edgetpu_panorama_object_detector/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /edgetpu_panorama_object_detector/parameter_updates [dynamic_reconfigure/Config]
 * /rosout [rosgraph_msgs/Log]

Subscriptions: 
 * /dual_fisheye_to_panorama/quater/output [sensor_msgs/Image]

Services: 
 * /edgetpu_object_detector/set_parameters
 * /edgetpu_object_node_manager/get_loggers
 * /edgetpu_object_node_manager/list
 * /edgetpu_object_node_manager/set_logger_level
 * /edgetpu_object_node_manager/start
 * /edgetpu_object_node_manager/stop
 * /edgetpu_panorama_object_detector/set_parameters

contacting node http://fetch15:37505/ ...
Pid: 15911
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound (37017 - 127.0.0.1:50484) [14]
    * transport: TCPROS
 * topic: /diagnostics
    * to: /diag_agg
    * direction: outbound (37017 - 127.0.0.1:51354) [16]
    * transport: TCPROS
 * topic: /edgetpu_panorama_object_detector/output/rects
    * to: /edgetpu_panorama_object_detection_rect_array_in_panorama_to_bounding_box_array
    * direction: outbound (37017 - 133.11.216.34:38024) [55]
    * transport: TCPROS
 * topic: /edgetpu_panorama_object_detector/output/class
    * to: /edgetpu_panorama_object_detection_rect_array_in_panorama_to_bounding_box_array
    * direction: outbound (37017 - 133.11.216.34:38022) [66]
    * transport: TCPROS
 * topic: /dual_fisheye_to_panorama/quater/output
    * to: /dual_fisheye_to_panorama/panorama_downsample_quater (http://fetch15:36483/)
    * direction: inbound
    * transport: TCPROS
```

On the other hand, `/edgetpu_panorama_object_detector/output/image` is published.

```
fetch@fetch15:~/ros/melodic/src/jsk-ros-pkg/coral_usb_ros$ rostopic hz /edgetpu_panorama_object_detector/output/image
subscribed to [/edgetpu_panorama_object_detector/output/image]
average rate: 10.585
	min: 0.057s max: 0.101s std dev: 0.01402s window: 9
average rate: 10.247
	min: 0.057s max: 0.101s std dev: 0.00975s window: 19
```

sktometometo, 2022-06-07:
僕がpanoramaに切り替えて使ってます。

708yamaguchi, 2022-06-07:
なるほど

ちなみにどうやって切り替えたらいいんですか

knorth55, 2022-06-07:
You can switch the object_detector by rosservice.
Someone change the object_detector from normal to panorama and leave it.

```
fetch@fetch1075:~$ rosservice call /edgetpu_object_node_manager/list
node_names:
  - edgetpu_object_detector
  - edgetpu_panorama_object_detector

fetch@fetch1075:~$ rosservice call /edgetpu_object_node_manager/start "name: 'edgetpu_object_detector'"
```

sktometometo, 2022-06-07:
今現在使ってます

tkmtnt7000, 2022-06-07:
https://github.com/knorth55/jsk_robot/pull/242#issuecomment-1139210350
この辺りの変更ですかね

708yamaguchi, 2022-06-07:
Thank you for your advices!
Very helpful.

closed 2022-06-07 by 708yamaguchi