jsk-ros-pkg/coral_usb_ros #116: Add some functions for own purposes

Petros626, 2022-07-06:
Hey @knorth55,

I finally reach the last step of my thesis and want to add some functions to the node, which I've found on other examples and from the pycoral API. So my approach is to implement the following: FPS counter (printed in window corner), execution time of the script and the detection results for documentation with pycoral.

the question is now can i just add it in the object_detector.py script or does it have to be added somewhere else in the node. that it is possible and has nothing directly to do with the rviz plug-ins i will assume.

Would be glad about a detailed answer, so that others can expand the node for their own purposes :)

FPS counter:
```
frame_rate_calc = 1
freq = cv2.getTickFrequency()
t1 = cv2.getTickCount()
cv2.putText(frame,'FPS: {0:.2f}'.format(frame_rate_calc),(30,50),cv2.FONT_HERSHEY_SIMPLEX,1,(255,255,0),2,cv2.LINE_AA)
t2 = cv2.getTickCount()
time1 = (t2-t1)/freq
frame_rate_calc= 1/time1
```

example execution time:
```
import time
start_time = time.time()
main()
print("--- %s seconds ---" % (time.time() - start_time))
```
get results from model:
```
pycoral.adapters.detect.get_objects(interpreter, score_threshold=- inf, image_scale=1.0, 1.0)
pycoral.adapters.detect.Object(id, score, bbox)
```

knorth55, 2022-07-06:
you can add the fps counter on the visualization image,
but we have `rostopic hz`, so it seems not exciting to me.
If you want to write the fps counter on visualization image, please change code below.
https://github.com/jsk-ros-pkg/coral_usb_ros/blob/2c5ddeb033a094795db927c31685fd7958104f8f/python/coral_usb/detector_base.py#L146-L194

Petros626, 2022-07-06:
yes I didn't think about the `rostopic hz` will pay attention to that. 
Okay so in `detector_base.py` everything personal could be add, thats I'm searching for :)

knorth55, 2022-07-06:
we don't have `iou_thresh`. we have `score_thresh`.

Petros626, 2022-07-06:
yes right, where i can config it? to change the score_thresh?

knorth55, 2022-07-06:
`score_thresh` can be changed from `dynamic reconfigure`.

https://github.com/jsk-ros-pkg/coral_usb_ros/blob/master/python/coral_usb/detector_base.py#L57-L66
https://github.com/jsk-ros-pkg/coral_usb_ros/blob/master/cfg/EdgeTPUObjectDetector.cfg#L9

closed 2022-07-12 by knorth55