jsk-ros-pkg/coral_usb_ros #117: run two models in one EdgeTPU

k-okada, 2022-07-07:
We'd like to know if two model runs in one EdgeTPU, please try it @knorth55

knorth55, 2022-07-07:
yes we can. we can run two models on Edge TPU.
But it depends on the network size and edgetpu memory size.
You cannot run multiple huge networks on edgetpu simaltaneously.

knorth55, 2022-07-07:
I tried to run `edgetpu_object_detector` and `edgetpu_panoram_object_detector` simaltaneously, but it does not work.

```
$ roslaunch coral_usb edgetpu_object_detector.launch INPUT_IMAGE:=/usb_cam/image_raw
... logging to /home/knorth55/.ros/log/0cb4a3de-fdbc-11ec-960d-507b9d9efafc/roslaunch-melodic-p50-22604.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/knorth55/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://melodic-p50:36071/

SUMMARY
========

PARAMETERS
 * /edgetpu_object_detector/device_id: 0
 * /edgetpu_object_detector/image_transport: raw
 * /edgetpu_object_detector/label_file: package://coral_u...
 * /edgetpu_object_detector/model_file: package://coral_u...
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    edgetpu_object_detector (coral_usb/edgetpu_object_detector.py)

ROS_MASTER_URI=http://localhost:11311

process[edgetpu_object_detector-1]: started with pid [22626]
[INFO] [1657174514.253881]: Using transport raw
[WARN] [1657174519.255549]: [/edgetpu_object_detector] subscribes topics only with child subscribers. Set '~always_subscribe' as True to have it subscribe always.
F :1133] HandleQueuedBulkIn transfer in failed. Not found: USB transfer error 5 [LibUsbDataInCallback]
[edgetpu_object_detector-1] process has died [pid 22626, exit code -6, cmd /home/knorth55/ros/melodic/devel/lib/coral_usb/edgetpu_object_detector.py ~input:=/usb_cam/image_raw __name:=edgetpu_object_detector __log:=/home/knorth55/.ros/log/0cb4a3de-fdbc-11ec-960d-507b9d9efafc/edgetpu_object_detector-1.log].
log file: /home/knorth55/.ros/log/0cb4a3de-fdbc-11ec-960d-507b9d9efafc/edgetpu_object_detector-1*.log
[edgetpu_object_detector-1] restarting process
process[edgetpu_object_detector-1]: started with pid [22767]
[INFO] [1657174532.650203]: Using transport raw
F :1133] HandleQueuedBulkIn transfer in failed. Not found: USB transfer error 5 [LibUsbDataInCallback]
[edgetpu_object_detector-1] process has died [pid 22767, exit code -6, cmd /home/knorth55/ros/melodic/devel/lib/coral_usb/edgetpu_object_detector.py ~input:=/usb_cam/image_raw __name:=edgetpu_object_detector __log:=/home/knorth55/.ros/log/0cb4a3de-fdbc-11ec-960d-507b9d9efafc/edgetpu_object_detector-1.log].
log file: /home/knorth55/.ros/log/0cb4a3de-fdbc-11ec-960d-507b9d9efafc/edgetpu_object_detector-1*.log
[edgetpu_object_detector-1] restarting process
process[edgetpu_object_detector-1]: started with pid [23483]
[INFO] [1657174538.655037]: Using transport raw
F :1133] HandleQueuedBulkIn transfer in failed. Not found: USB transfer error 5 [LibUsbDataInCallback]
[edgetpu_object_detector-1] process has died [pid 23483, exit code -6, cmd /home/knorth55/ros/melodic/devel/lib/coral_usb/edgetpu_object_detector.py ~input:=/usb_cam/image_raw __name:=edgetpu_object_detector __log:=/home/knorth55/.ros/log/0cb4a3de-fdbc-11ec-960d-507b9d9efafc/edgetpu_object_detector-1.log].
log file: /home/knorth55/.ros/log/0cb4a3de-fdbc-11ec-960d-507b9d9efafc/edgetpu_object_detector-1*.log
```

closed 2022-07-19 by knorth55