jsk-ros-pkg/coral_usb_ros #120: Project name in CMakeLists.txt is different from package name?

Petros626, 2022-07-21:
Package name `coral_usb_ros` and project name in CMakeLists.txt `project(coral_usb)` is different which is unusual. It makes more confusing to run catkin for single package like: `catkin_make --only-pkg-with-deps coral_usb` because you need to check CMakeLists.txt to find project name

https://github.com/jsk-ros-pkg/coral_usb_ros/blob/2c5ddeb033a094795db927c31685fd7958104f8f/CMakeLists.txt#L2

knorth55, 2022-07-21:
Yes, I know the problem.
I think the repo name should have ros in it, to emphasize it is ros package.
But package name should not have ros in it, because when tou release the package, the package name would be ros-melodic-coral-usb-ros, which has two ros!

knorth55, 2022-07-21:
This is just a execuse, but sorry.

Petros626, 2022-07-21:
In that case maybe it would be a better solution to change the package name during git clone, using: 

`git clone https://github.com/jsk-ros-pkg/coral_usb_ros.git coral_usb`
https://github.com/jsk-ros-pkg/coral_usb_ros/blob/2c5ddeb033a094795db927c31685fd7958104f8f/README.md?plain=1#L162
https://github.com/jsk-ros-pkg/coral_usb_ros/blob/2c5ddeb033a094795db927c31685fd7958104f8f/README.md?plain=1#L183
https://github.com/jsk-ros-pkg/coral_usb_ros/blob/2c5ddeb033a094795db927c31685fd7958104f8f/README.md?plain=1#L201

closed 2022-08-01 by Petros626