jsk-ros-pkg/coral_usb_ros #121: node publishes faster than receiving messages?

Petros626, 2022-07-25:
Hey,

I experienced the weird behaviour, that the topic `/output/image` is published faster than receiving the topic `/sensor_msgs/Image`. I thought the incoming message rate would specify how fast the image is published?
Any idea if it belongs to the node?

Input: Output
sensor/msgs_Image (every 2 sec.) -> coral usb node -> edgetpu_object_detector/output/image (every 0.5/1 sec.)

knorth55, 2022-07-26:
https://github.com/jsk-ros-pkg/coral_usb_ros#parameters

`visualization` loop is anothr loop.
we can control the hz by  `visualization_duration`.

Petros626, 2022-07-26:
> https://github.com/jsk-ros-pkg/coral_usb_ros#parameters
> 
> `visualization` loop is anothr loop. we can control the hz by `visualization_duration`.

the visualization frequency in rviz is the same the messages are coming (2 Hz). By using using `rostopic echo /edgetpu_object_detector/output/image` I noticed the difference from 2Hz to >2Hz.

Your suggestions works good with a value of `2.0`, but it wonders me, how the node could `echo` the pixel values faster than the incoming `Image message`...

knorth55, 2022-07-28:
> Your suggestions works good with a value of 2.0, but it wonders me, how the node could echo the pixel values faster than the incoming Image message...

The node just keep the old image and use it to publish in >2Hz.

Petros626, 2022-07-28:
Yes, but your suggestion helps, will close the issue

closed 2022-07-28 by Petros626